prusa3d/PrusaSlicer #677: Negative Feedrates: G1 F-3.84007e+006  stops smoothie

pflugj, 2018-01-08:
[15mm 4x4 Frame&Lid.zip](https://github.com/prusa3d/Slic3r/files/1610674/15mm.4x4.Frame.Lid.zip)

[config.txt](https://github.com/prusa3d/Slic3r/files/1610672/config.txt)

### Version
_Version of Slic3r used goes here_
1.38.6 prusa3d win64

### Operating system type + version
Win 10

### Behavior
Negative feed rate stops Smoothie.
Octoprint reports:
Feedrate <0

* _Actual Results_
Negative Feedrates in G-Code like:
G1 F-3.84007e+006

_Is this a new feature request?_
NO

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-02-02:
Fixed with https://github.com/prusa3d/Slic3r/commit/ea9920e5d679e2deb19e553ee504cafea7b89714

The problem was caused by the gap fill algorithm, which worked
with square extrusion width values as with rounded extrusion widths,
which sometimes lead to negative extrusion cross sections 
for high height to width ratios.

The extrusion width logic has been changed to consider the input width
to be the extrusion spacing, not the extrusion width. The change certainly
removed the negative feed rates, but it also certainly increased the gap
fill width to some exent. It needs to be verified now, whether the gap fill
does not extrude too much.

We will test the fix internally before closing this issue.

bubnikv, 2018-02-12:
Fixed, closing.

closed 2018-02-12 by bubnikv