prusa3d/PrusaSlicer #678: In individual objects mode, brims not printed individually

chambm, 2018-01-08:
### Version
1.36.2

### Operating system type + version
Windows 7 64-bit

### Behavior
When I print a bunch of small parts using individual objects mode, it seems odd that slic3r prints all the brims first, then starts filling in the brims individually. 

* _Expected Results_
Wouldn't there be better brim adhesion if it started printing the object right after its brim? This would be consistent with previous fixes to print the first layer from the outside in.

* _Actual Results_
What happens now is first the skirt is printed, then each brim is printed, then it starts printing the first object (usually this is the last brim printed).

#### STL/Config (.ZIP) where problem occurs
[cube-gears-2x-7pins.zip](https://github.com/prusa3d/Slic3r/files/1612389/cube-gears-2x-7pins.zip)

x0rtrunks, 2019-04-22:
Are there still plans to implement this?

I like to use a single brim layer as a nozzle prime right before the part starts printing. Much like the skirt does before printing the first layer. By printing the brim before each part, you're able to guarantee a good nozzle prime after the travel move between parts.

poge-st, 2020-11-25:
PrusaSlicer: version 2.2.0 
Problem is still present 

Attached an image of a print which is jsut running at this precise moment ;)
[image]

rtyr, 2021-01-20:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/292. Closing this one.

closed 2021-01-20 by rtyr