prusa3d/PrusaSlicer #679: Buiild: missing tbb lib

supermerill, 2018-01-08:
### Version
_Version of Slic3r used goes here_
last stable on git at early january 2°17
b93d26517d16263e95ad7dcdb6880a75cc2a7632

### Operating system type + version
windows 10 x64

### Behavior
when compiling on windows, following the cmke wiki, i have to add 
` C:\local\tbb\lib\ia32\vc12\tbb_static.lib C:\local\tbb\lib\ia32\vc12\tbbmalloc_static.lib C:\local\tbb\lib\ia32\vc12\tbbmalloc_proxy_static.lib`
to the file build/xs/CMakeFiles/XS.dir/build.make , at the last line for the linking of XS.dll (line 252~)

Maybe something is missing...

bubnikv, 2018-02-12:
Thanks for the note.
It is the TBB_INCLUDE_DIR and TBB_LIBRARY missing in the batch file packed with our perl binary package. I shall fix that.

dartrax, 2018-02-16:
That would be nice since I just came across the same problem. Thanks for a solution, @supermerill.

dartrax, 2018-02-17:
In fact, I needed to add

> set(TBB_INCLUDE_DIR "C:/local/tbb/include")
> set(TBB_LIBRARY "C:/local/tbb/lib/**intel64**/vc12")

to the CMakeLists.txt

Otherwise, I would get a fatal error at 98% of `nmake`...

> **tbb_static.lib(tbb_misc.obj) : fatal error LNK1112: module machine type 'X86' conflicts with target machine type 'x64'**
> NMAKE : fatal error U1077: ""C:\Program Files\CMake\bin\cmake.exe"": Rückgabe-Code "0xffffffff"
> Stop.
> NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"": Rückgabe-Code "0x2"
> Stop.
> NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"": Rückgabe-Code "0x2"
> Stop.

vojtechkral, 2018-02-22:
FWIW, you don't need to set the variables inside `CMakeLists.txt` if you setup the `TBB_INSTALL_DIR` environemtn variable to point to the TBB directory. We'll update the cmdline scripts to reflect this.

dartrax, 2018-02-23:
I tried that first since I hesitated touching the `CMakeLists.txt`, but that did not work - I don't know why. May be I made something wrong.

closed 2018-03-13 by supermerill