prusa3d/PrusaSlicer #680: Pronterface Not Opening MMU Single Mode Files

Ziethriel, 2018-01-09:
Version:
1.38.6

Operating system type + version:
Windows 10

Behavior:
Describe the problem:
Multi Material Single Mode gcodes don't open in pronterface
Steps needed to reproduce the problem:
1.Slice Model with multi material single mode
2.Open with pronterface

Expected Results:
File opens and relays an estimated print time
Actual Results
File does not open.

Is this a new feature request?
No

PS. I expect this is because a nozzle is selected when you print instead of in the slicer.

bubnikv, 2018-02-12:
Will it open in pronterface, if you manually replace the T? line with T0 in the G-code?

bubnikv, 2018-02-13:
I have created a pronterface ticket to support the T? G-code.

https://github.com/kliment/Printrun/issues/888

Closing.

closed 2018-02-13 by bubnikv