prusa3d/PrusaSlicer #684: Full Soluble Materials interfacing into Models + Soluble material noise on Bed

MakingThingsWorkToronto, 2018-01-11:
### Version
1.37.2-prusa3d-win64

### Operating system type + version
Windows 10 Fall Creators Update

### Behavior / Steps to reproduce
Add several Zaribo models to bed: http://zaribo.org/blog/stl/ (see attached STLs below).
Organize parts
Select the "soluble full" preset.
Slice
Preview
Note following issues:

1) Soluble material interfacing into model on bed such that when dissolved there will be holes in the model or overlaps between materials.

https://spgoo-my.sharepoint.com/personal/matt_makingthings_work/_layouts/15/guestaccess.aspx?docid=07426a48f8892440c92004a1390805983&authkey=AdNYg5T0Qg_uDZDjkO3kuTA&e=nNYBK1

[image]

2) Soluble material seemingly randomly printed on bed with patterns that interfere with things like brims & potentially other parts.

https://spgoo-my.sharepoint.com/personal/matt_makingthings_work/_layouts/15/guestaccess.aspx?docid=076c62b2bac814f429d6770b1b5232c5b&authkey=AcvzVnS9btj30WQnR4M3vhY&e=pWClc0

[image]

I wouldn't expect so much noise on the bed.  The more plastic extruded the greater the chances of something going wrong.

I would expect that when a brim is added to parts the soluble material would if necessary be printed on top of the brim.

#### STL/Config (.ZIP) where problem occurs
Please see above public link to STLs as well as the attached which were optimized for printing.
[STL.zip](https://github.com/prusa3d/Slic3r/files/1624200/STL.zip)

bubnikv, 2018-01-12:
I believe this is a duplicate of https://github.com/prusa3d/Slic3r/issues/316 , isn't it?

MakingThingsWorkToronto, 2018-01-13:
The first part of the bug looks like a duplicate.  The noise on bed isn't a big issue but potentially related? I'm cool if this is a duplicate.

MakingThingsWorkToronto, 2018-01-13:
Having said that the issue is significant. I've had parts fail because of the support material interfacing into the print.  At best it's unreliable & in practice unusable.  A bit shocking this functionality is promoted when it is so imperfect.

bubnikv, 2021-02-24:
Fixed with fcb714cd241f0847522b9aac57da6b62acdd0152

Changed the 1st layer infill to rectilinear even for soluble materials.

New parameters raft_first_layer_density and raft_first_layer_expansion
to influence the 1st layer of raft or support.

closed 2021-02-24 by bubnikv