prusa3d/PrusaSlicer #685: Whining Sound When 2nd Layer Starts

tbedore, 2018-01-11:
### Version
1.38.7-prusa3d

### Operating system type + version
macOS High Sierra - 10.3.2

### Behavior
* When I would start printing, everything would work fine, and then layer 2 it would start printing a little, and then it would stop making a loud whining sound and would just be stuck on the layer, not doing anything, the read-out on the LCD was normal.  
* I sliced something in Slic3r and in order to see the preview, you need to hit slice now.  So you can slice anything in the newest drivers, hit slice now, and then export the g-code.  That is all I did, and it failed on the 2nd layer, when I went to go check it, the printer was running at 235 and it was cooling down from a 250 temp.  
* The printer should print and continue the whole print, and then finish it.  
* It fails on a part of layer 2, ONLY IN SLIC3R
 

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/1624388/Archive.zip)

bubnikv, 2018-02-12:
Which printer?

tbedore, 2018-02-12:
It’s the MK2s

bubnikv, 2018-02-12:
Would you please send me your G-code that failed?

Is it possible, that the whining sound is due to one of the pulleys slipping? Where does the whining noise come from?

bubnikv, 2018-03-14:
We did not receive a response in a month, in particular we did not receive the G-code to test on our printers. Closing.

closed 2018-03-14 by bubnikv

tbedore, 2018-03-14:
Sorry, I never got an email about your previous response, I just saw these.  I don't have the gcode file anymore for it, and the sound was not coming from any pulleys or anything, it was coming from the buzzer in the LCD.

tbedore, 2018-03-14:
But in addition, since the newest drivers, I haven't had these issues since.

bubnikv, 2018-03-14:
Great to know the issue solved itself :-)