prusa3d/PrusaSlicer #686: Crash while trying to slice with a raft

Sylvian, 2018-01-13:
### Version
1.38.6-prusa3d-win64 (MK2 Multi Material)

### Operating system type + version
Windows 10 (64bits)

### Behavior
Crash while slicing object

Steps to reproduce:
- Add a STL file into Slic3r
- Choose "0.20mm NORMAL" as print settings
- Choose "Original Prusa i3 MK2 MultiMaterial" as the printer
- Set Filament 1 => Generic PLA and Filament 2 => Polymaker PC-Max
- Double click on the item to print and set "Extruder" to 2 (to make it in PC-Max)
- Go to "Print Settings" tabs
- Set "Raft layers" to 3 into "Support material"
- Set "Support material/raft/skirt extruder" to 1 into "Multiple Extruders"
- Set "Support matieral/raft interface extruder" to 1 into "Multiple Extruders"
- Go back to the "Plater" tab and click "Slice now"

Then you got a software crash, I just try to make a gcode file with a PLA raft and a PC item, maybe am I making something wrong ?

Thanks for your help.

(STL File I'm using is available here: https://puu.sh/z0FWP/da41920884.stl)

fpsoares, 2018-01-24:
Same issue here, MM prints just fine with no raft or just brim, but crashes when slicing raft every time.

pleedom, 2018-02-09:
Same problem here, just tried alpha 1.39.1, cant slice with a raft =\

bubnikv, 2018-02-12:
@Sylvian I was able to reproduce your issue and to fix it. It will be in the upcoming beta.

@fpsoares, @pleedom
Would you please provide your data to reproduce your issues? I wonder whether your issues are the same as of @Sylvian. Thanks.

Sylvian, 2018-02-12:
Thanks for the fix, hope to get it ASAP ^^

fpsoares, 2018-02-12:
I believe issue is the same as Sylvian. Thanks for the fix

bubnikv, 2018-02-12:
The issue of @Sylvian  is specific to a MM print with raft printed with one material, the object with another material, and no support.

@fpsoares, @pleedom Is this your scenario as well?

fpsoares, 2018-02-12:
That is correct, I had not tried woth supports as my model did not require
it, but adding supports will cause the model to slice with no issues

pleedom, 2018-02-12:
@bubnikv I didn't try changing the support material itself, I only experience this issue using more than one selected material and using a raft when "Generate Supports" is unchecked. I did not get the crash when using single material mode.

bubnikv, 2018-02-13:
Fixed with https://github.com/prusa3d/Slic3r/commit/a5e48cdf4d824fdec8a97d1f1569e58595d0990e.
Thanks for the report. Closing.

closed 2018-02-13 by bubnikv