prusa3d/PrusaSlicer #687: X Window System error

alpharesearch, 2018-01-13:
### Version
_Version of Slic3r used goes here_
Slic3r-1.38.5
_Use `About->About Slic3r` for release versions_
can not start it
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
./Slic3r-1.38.5-prusa3d-linux64-full-201712212010.AppImage
### Operating system type + version
_What OS are you using, and state any version #s_
Mint 18.3 Mate
### Behavior
* _Describe the problem_
I type:
./Slic3r-1.38.5-prusa3d-linux64-full-201712212010.AppImage
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 558 error_code 2 request_code 154 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

and get this back... if I type wx-config --version
3.0.2
I get this
* _Steps needed to reproduce the problem_
It happens with that appimage and with the git with full build.
  * _If this is a command-line slicing issue, include the options used_
na
* _Expected Results_
start the gui
* _Actual Results_
stops error message
  * _Screenshots from __*Slic3r*__ preview are preferred_
na

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
na

bubnikv, 2018-02-12:
May it be something similar to https://github.com/prusa3d/Slic3r/issues/182 ?

It sounds like the libraries bundled with the Slic3r AppImage are not compatible with your distro. I bet if you compile Slic3r yourself, it will work.

Do you know, whether you will be able to use debian or ubuntu packages on Mint? I know there is an official package of our Slic3r for Debian.

https://packages.debian.org/sid/slic3r-prusa

alpharesearch, 2018-02-12:
Thanks for looking at this.

I did compile today (after getting the lastest master) and last month as well, I got the same error on both the AppImage and the compile in the past. Today, I only did compile and I'm getting this now:
> "sanitize_identifier" is not exported by the Sub::Quote module
> Can't continue after import errors at /home/github/prusa/Slic3r/local-lib/lib/perl5/Method/Generate/Accessor.pm line 7.
> BEGIN failed--compilation aborted at /home/github/prusa/Slic3r/local-lib/lib/perl5/Method/Generate/Accessor.pm line 7.
> Compilation failed in require at /home/github/prusa/Slic3r/local-lib/lib/perl5/Moo.pm line 161.
> Compilation failed in require at /home/github/prusa/Slic3r/local-lib/lib/perl5/Moo.pm line 173.
> Compilation failed in require at /home/github/prusa/Slic3r/lib/Slic3r.pm line 55.
> BEGIN failed--compilation aborted at /home/github/prusa/Slic3r/lib/Slic3r.pm line 55.
> Compilation failed in require at ./slic3r.pl line 16.
> BEGIN failed--compilation aborted at ./slic3r.pl line 16.

Looks like something is not right on this host PC, any idea? Because if I do everything in a clean Mint 18.3 virtual machine it is working? Are there any packages or directories that I should reinstall or delete?

alpharesearch, 2018-02-12:
I can now run the new AppImage on both the host and guest. I did delete the git directory on the host and started compiling again and now everything is working... Mint is based on Ubuntu so I guess some package was updated last month that caused the issue. I remember seeing a new Nvidia graphics card driver being installed, but I'm not sure if this was the fix.

bubnikv, 2018-02-12:
Great news, Markus. I am happily closing this issue :-)

closed 2018-02-12 by bubnikv