prusa3d/PrusaSlicer #688: Value expression not functioning in dual extruder setup

freiser77, 2018-01-14:
### Version
1.38.5-prusa3d

### Operating system type + version
macOS High Sierra

### Behavior
* I want to lower the temperature of the unused extruder on tool change. So I insert to following snippet in the tool change g-code section:

<pre>
{if previous_extruder==0}
M104 S{temperature_0-45} T0; let unused extruder cool down
{else}
M104 S{temperature_1-45} T1; let unused extruder cool down
{endif}
</pre>
* Slicing works ok, on save I get the following error message:

<pre>
!!!!! Failed to process the custom G-code template toolchange_gcode
Parsing error at line 8: Not a variable name
M104 S{temperature_0-45} T0; let unused extruder cool down
       ^
!!!!! End of an error report for the custom G-code template toolchange_gcode
</pre>

Do I have a syntax error or why is it failing?

bubnikv, 2018-01-14:
Indexing of array variables referenced from inside the curly braces has the
usual format as in C or java script. The legacy syntax with the underscore
has been retained if used inside the square braces.

examples:
{array [1]}
is equivalent to
[array_1]

freiser77, 2018-01-14:
@bubnikv: Thanks.

closed 2018-01-14 by freiser77