prusa3d/PrusaSlicer #689: slic3r prusa edition 1.38.6 custom g-code

Deckill, 2018-01-16:
### Version
_Version of Slic3r used goes here_
1.38.5 and 6

### Operating system type + version
Windows 10

### Behavior
The custom g-code does not work.
when I type something it just comments out with ; at the very end.(on both filament gcode and tool change gcode on printer setting)
and when I use multi meterial, it automatically retract 92mm on tool change, causing filament stuck on mine.
can you help me with this?
I wish I can only retract 35mm OR disable it so I can manually type some cool gcode with it.

Thank you 

_Is this a new feature request?_

I think so

bubnikv, 2018-01-19:
please provide more detail on your script. please provide a config file
(export from menu file->export config). An stl may be useful as well.

bubnikv, 2018-03-14:
Didn't receive any feedback for a month. Closing.

closed 2018-03-14 by bubnikv