prusa3d/PrusaSlicer #690: modifier menu do not appear to change scale

hsuq510, 2018-01-16:
### Version
1.38.5

### Operating system type + version
MacOS 10.11.6

### Behavior
Context menu on modifier do no appear when clicking on it with right mouse button. Cant change scaling.

bubnikv, 2018-01-19:
is it a new issue? will a double click work? by the way, I did not know
there is a right button on macos mouse.

hsuq510, 2018-01-19:
neither double click or any other combination works. The menu do not appear.

bubnikv, 2018-05-10:
@FidelCapo Would you please test on Mac? Thanks.

bubnikv, 2018-05-16:
Sorry, but we cannot reproduce your issue. Do you still have it with
https://github.com/prusa3d/Slic3r/releases/tag/untagged-8fc8ca127d0924fe5504 ?

I am not an OSX expert, but don't you have your mouse configured in a special way? Left handed or something else?

bubnikv, 2018-05-16:
Maybe I understand now. Are you asking for the following menu?

[image]

This has never been implemented in Slic3r PE, only in the upstream slic3r.

Would you please confirm this?

bubnikv, 2018-08-02:
No response. We will merge the functionality for this dialog into the platter right panel, so the UI will be different anyway. Closing.

closed 2018-08-02 by bubnikv