prusa3d/PrusaSlicer #692: Mac installation has wrong permissions

larsrc, 2018-01-21:
### Version
1.38.4, 1.38.5 at least

### Operating system type + version
Mac OS X 10.12.6

### Behavior
* _Describe the problem_
  * When installing Slic3r on a Mac, it is only executable by the user installing it.
* _Steps needed to reproduce the problem_
  1 Install Slic3r
  2 Switch to a different user
  3 Open applications folder
* _Expected Results_
  * I can run Slic3r
* _Actual Results_
  * I don't have permissions to run Slic3r:

[image]

Getting the info on the Slic3r App on the dmg also shows that only the user mounting it has access.

bubnikv, 2018-02-02:
Do you have experience how to achieve your request? Shall I just change the Unix permissions of the files bundled inside the dmg archive?

larsrc, 2018-03-09:
I don't know first thing about dmg archives, sorry. I found that changing them after installation didn't help.

bubnikv, 2018-03-14:
Did you install other applications into the main Applications folder the same way as you did it with Slic3r?

Reading the following discussion, maybe you need to adjust the rights to the Slic3r.dmg file yourself after you moved it to the main Applications folder, so it will be accessible to everyone?

https://forums.macrumors.com/threads/can-i-install-and-application-and-have-it-available-for-all-users.170148/

bubnikv, 2018-07-21:
Also this may help:

https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/slic3r-file-permissions-problem-with-mac-install-t19660.html

bubnikv, 2018-07-24:
We are now copying the readable / executable attributes for the files inside the .dmg archive from the "owner" to the "group" and "others", so we hope the issue goes away. Interestingly enough, our tester @FidelCapo was not be able to reproduce the issue, so we will be thankful, if you could test the attached build for us.

[Slic3rPE-1.41.0-alpha2+22-full-ge0e6a23-201807241358.dmg.zip](https://github.com/prusa3d/Slic3r/files/2224105/Slic3rPE-1.41.0-alpha2%2B22-full-ge0e6a23-201807241358.dmg.zip)

bubnikv, 2018-07-27:
No response, but we hope we fixed the issue with the upcoming Slic3r 1.41.0-alpha3 release. Closing.

closed 2018-07-27 by bubnikv