prusa3d/PrusaSlicer #697: Slic£R behaves strangely with different nozzles sizes

jeremypearce, 2018-01-28:
Hi 

Version SLic3R Prusa Edition
Version 1.38.6-prusa3d-win64
Windows 10.

When using  nozzles different to 0.4 mm not all parameters are updated in the advanced section
They remain I believe as per 0.4 mm nozzle
Putting in 0 for default gives very bad behaviour.

Many thanks
Jeremy

Jeremy

### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2018-01-30:
You can set the fields. Put 1.05~1.1 * your nozzle size on each box.

bubnikv, 2018-02-12:
> When using nozzles different to 0.4 mm not all parameters are updated in the advanced section
> They remain I believe as per 0.4 mm nozzle
> Putting in 0 for default gives very bad behaviour.

I am sorry, but this is a too little information to me. Would you please provide examples?

bubnikv, 2018-03-14:
Too little information provided, no response in a month, closing.

closed 2018-03-14 by bubnikv