prusa3d/PrusaSlicer #698: Settings choice limited to 99.

rongith, 2018-01-29:
### Version
1.38.5

### Operating system type + version
Windows 10 x64

### Behavior
I have 124 .ini files in my print settings folder. I can only select 99 of them in the rolling menu.

bubnikv, 2018-03-14:
Would you please zip and attach the content of your Slic3rPE directory, or export and attach your config bundle? That would help us to test quickly. Thanks.

bubnikv, 2018-04-18:
@rongith By the way, do you get the same issue with 1.39.1? We changed this part of code from Perl to C++, so maybe your issue has already been solved.

rongith, 2018-04-18:
Sorry not to answer quickly to your first request. For now I don't have the use of 100+ print settings anymore as I have done some cleanup and I use to keep Prusa settings as legacy settings for reference but your settings has decreased a lot this year. But in a near future I will go again to more than 100 print settings.
I just checked I can now select all print settings that are in the folder when there are 100+.
Thank you.

closed 2018-04-18 by rongith