prusa3d/PrusaSlicer #699: Limited function of "Combine infill every:"

landway, 2018-01-30:
### Version
1.38.5
1.37.2

### Operating system type + version
windows 7
windows 10

### Behavior
Cannot be set effectively "Combine infill every" more than 2.

If set for example 4, than in visualization is shown only every 2.
Three layers should be omitted, but there is only 1.
[image]
[image]

supermerill, 2018-01-30:
Because your layer height can't be > your nozzle width.
I assume you use 0.2 layer height on a 0.4 nozzle?

rtyr, 2021-01-20:
I believe it was answered. Closing.

closed 2021-01-20 by rtyr