prusa3d/PrusaSlicer #700: Fan speed change for briding...?

Petclaud, 2018-01-30:
### Version
1.38.5

### Operating system type + version
Windows 10

### Behavior
* Printing the attached design. I have given it several attempts, all bad result of the small bridges

* Just try to print the attached GCODE. 

t.

* At a layer hight of 14.15 mm I would expect the printer to go to 'briding mode' when printing the small bridges, ie bridge speed, flow and cooler fan change. 

* What I can check on the display, is that the cooling fan does not increase speed, but instead keeps running on the same speed as the other layers of the print. I could not check the flow and speed, but it looked like the print speed increased (which might be explained because the peremiters are rather small and these are printed at 20mm/s and the bridges are set to print at 30mm/s, however one could also argue that these bridges are also like small perimeters, hence should also be printed at 20 mm/s...

[image]
[image]

  * _Screenshots from __*Slic3r*__ preview are preferred_

No new request

#### STL/Config (.ZIP) where problem occurs

[config.zip](https://github.com/prusa3d/Slic3r/files/1679400/config.zip)
[MagLacesSportLow_V4.zip](https://github.com/prusa3d/Slic3r/files/1679402/MagLacesSportLow_V4.zip)

bubnikv, 2018-05-10:
Did you check your G-code for the short bridges?

My best bet is that the G-code is there, but your fan does not kick in quickly enough.

Petclaud, 2018-05-10:
Thx Bubnikv. The gcode was already attached in the initial post. I have tried to find M106 S... command at Z14.15 but cannot find it. I ran the code through GCODE ANALYZER and I see that the initial bridge-lines at Z height 14.15 run at the Perimeter Speed setting (40mm/s) instead of Bridge speed (30mm/s)

[image]

I see that I had not attached the STL; here now attached as well. 

[MagLacesSportLow_V4-STL.zip](https://github.com/prusa3d/Slic3r/files/1992833/MagLacesSportLow_V4-STL.zip)

closed 2018-07-25 by Petclaud