prusa3d/PrusaSlicer #701: Not Generating Supports at All

christizzz, 2018-01-30:
1.38.5 Win64

Windows 10 Home Edition(64Bit)

### Behavior
* Supports Does Not Generate at all when in in Preview
* I load an STL, activate Support, Slice it... i get a loading generating support but nothing shows under Preview.

supermerill, 2018-01-31:
Can't reproduce

christizzz, 2018-01-31:
ok is there a way to completely reset Slic3r ? maybe i have a faulty installation...

bubnikv, 2018-01-31:
Slic3r 1.38.5 and newer has the option to run the wizard from the help menu
and replace all the settings with the stock ones.

christizzz, 2018-01-31:
O i run the wizard again... choose "other" as printer...i built a prusa from reprap.
I imported my config... still no supports.... here are my configs ( per default supporet is inactive i need to activate )

[Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/1682627/Slic3r_config_bundle.zip)

bubnikv, 2018-02-12:
also please provide your model and screenshots.

bubnikv, 2018-03-14:
No response for a month, closing.

closed 2018-03-14 by bubnikv