prusa3d/PrusaSlicer #702: Interior brim option

hoegge, 2018-01-31:
It would be very useful to have the "interior brim" options from Slic3r in Slic3r PE also.

[https://github.com/alexrj/Slic3r/issues/2026](url)

Thanks

supermerill, 2018-01-31:
My advice: if you need a special brim, you can embedded it into your model (or a companion one). With that, you can also say goodby to good ol' brim and say hello to "mickeymouse ears" that are much easier to remove and stickier.

hoegge, 2018-02-03:
Well - knowing that trick already, I do prefer the ability to make internal brim too, when you have quick turn-arounds, and just want to print your STL as it is, whithout having to do special modelling for printing (e.g. ears)

bubnikv, 2018-02-12:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/491 and https://github.com/prusa3d/Slic3r/issues/61. Closing this issue.

closed 2018-02-12 by bubnikv