prusa3d/PrusaSlicer #703: Unsupported top layers causing sagging holes

comps, 2018-02-01:
### Version
1.38.5-prusa3d-linux64 (binary tarball from github)

### Operating system type + version
Linux (Fedora)

### Behavior
Models with solid interior (turned into infill) do not properly support additional solid structures on the top surface.

[image]

As the extruded paths are supported only on one end, this causes them to sag downwards, creating holes aligned to the infill pattern below. This pattern split happens with any infill pattern and any infill percentage, even 100% where a separate infill pattern is used (using infill extrusion width instead of top surface extrusion width), but is most noticeable with smaller infill percentages.

Slic3r should either do a full top surface everywhere and lay the structures on top on it, or otherwise calculate contact points of the underlying infil so it always does bridges supported on both sides and builds any additional perimeters/infill on top of the supported bridges.

#### STL/Config (.ZIP) where problem occurs
[config.txt](https://github.com/prusa3d/Slic3r/files/1686783/config.txt)
[stl-with-gcode.zip](https://github.com/prusa3d/Slic3r/files/1686787/stl-with-gcode.zip)

comps, 2018-02-12:
It should be noted that this bug is shared by at least Slic3r Prusa Edition, the original Slic3r (nightly builds), Lulzbot Cura and Ultimaker Cura 3.1, where both Cura editions actually position the perimeters at the edges of the hole, not some 1-2mm away like Slic3rPE (making it worse than any Slic3r edition).

A quick and IMHO fairly simple fix would be a toggle like "Always fill in top surfaces", which would always do full bridges over any infill before building on top - on some (mostly vertical) models, disabling it would save more plastic, but on models like mine (above), it would greatly boost structural integrity.

bubnikv, 2018-02-12:
I believe this is a duplicate of https://github.com/prusa3d/Slic3r/issues/569. Please continue your discussion at https://github.com/prusa3d/Slic3r/issues/569.

closed 2018-02-12 by bubnikv