prusa3d/PrusaSlicer #704: Exception in lib/site_perl/5.22.0/Slic3r/Print/Objext.pm line 66

aligator, 2018-02-03:
### Version
1.39.1-alpha-prusa3d-linux64

### Operating system type + version
Linux Mint x64

### Behavior
when slicing this thing: 
https://www.thingiverse.com/thing:2081602
I get an error:
`Caught C++ exception of unknown type at [...]/lib/site_perl/5.22.0/Slic3r/Print/Object.pm line 66`

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/prusa3d/Slic3r/files/1691722/config.zip)

bubnikv, 2018-02-12:
Thanks, I see the exception on Linux, I don't get this exception on Windows.

bubnikv, 2018-02-12:
The exception was throw by the Clipper library. The behavior with regard to this issue was different between Windows and Linux due to a different interpretation of the C++ long type on 64bit systems:

https://software.intel.com/en-us/articles/size-of-long-integer-type-on-different-architecture-and-os

To match the behavior of Slic3r on Windows and Linux, I changed the coordinate type from long to int32_t , and now your model can be sliced on Linux as well.

So let's call it fixed with 47d904a628ec35a1a0f08ab2ad7bd664cc0c601f.

closed 2018-02-12 by bubnikv