prusa3d/PrusaSlicer #706: Export STL, Slice now, and Export gcode buttons are cut off

dmeehl, 2018-02-03:
### Version
Slic3r-1.38.5-prusa3d-linux64

### Operating system type + version
KDE Neon User Edition 5.11

### Behavior
[Export STL], [Slice now], and [Export gcode] buttons are cut off. There is not enough vertical space to read the text in the buttons. See attached screenshot.
[image]

#### STL/Config (.ZIP) where problem occurs
This is not a problem with slicing or model. Strictly UI.

bubnikv, 2018-02-12:
This issue sounds similar to https://github.com/prusa3d/Slic3r/issues/85
@VojtechKral will look into it.

bubnikv, 2018-05-10:
@vojtechkral @YuSanka any update on this issue?

YuSanka, 2018-05-13:
@dmeehl , Thanks a lot for your report. Fixed bug is merged in the current master, next public release is expected in several days. Please, give us feedback after you have the chance to test it.

bubnikv, 2018-05-14:
Fixed in master, an alpha will be released soon.

closed 2018-05-14 by bubnikv