prusa3d/PrusaSlicer #709: Faulty usage of XY separation value

EddyMI3d, 2018-02-05:
### Version
1.38.6-prusa3d-win64

### Operating system type + version
Windows 7 Home Premium 64
Service Pack 1

### Behavior
In Tab "Print Settings"
Sub menu "Support material"
Menu point "XY separation between an object and its support:"
The "%" sign of a value is ignored and the value is taken as "mm".

bubnikv, 2018-02-12:
Are you sure? I find this highly improbable, as we use 60% of the external perimeter width and Slic3r certainly does not make the gap 60mm wide.

EddyMI3d, 2018-02-12:
Sure, the 60% are fine, but when you change the value, you can't get back to 60% because slicer interprets "60%" as "60 mm"

EddyMI3d, 2018-02-12:
Addendum: when I change to 60%, then save the profile, quit slic3r and restart, the 60% is accepted

bubnikv, 2018-02-12:
I think the difference here is, you are entering a space between the number
and the percent sign. If you type the percent sign just after the number,
it will work.
I will implement some workaround into slic3r to accept the space.

EddyMI3d, 2018-02-12:
No, there is no space. And if there is one, it's deleted by Slic3r.

bubnikv, 2018-02-12:
I think I found the bug. It is a regression issue since the 1.38.x series,
where the configuration layer has been reworked. There is a bug, where
Slic3r considers the value with and without percent equivalent, therefore
once it accepted the number without percent, it will ignore the percent
sign once added.

I will fix it for the 1.39.1 final.

EddyMI3d, 2018-02-12:
Thanks a lot for your efforts!

bubnikv, 2018-02-12:
Fixed with https://github.com/prusa3d/Slic3r/commit/6f92424bab82f1795697b21e074e4652529271ff
Thanks for the report.

closed 2018-02-12 by bubnikv