prusa3d/PrusaSlicer #711: Error when using Windows Paths for drives.

eddie-moser, 2018-02-07:
### Version
1.38.6
### Operating system type + version
Windows 10

### Behavior
When installing Slic3r into a folder that is actually a drive mounted as a windows path it will not open several images keeping people from being able to save new profiles etc...

_Is this a new feature request? no

bubnikv, 2018-02-12:
What exactly does not work? I tried to extract Slic3r onto a network drive and at the first sight everything seems to be working.

eddie-moser, 2018-02-12:
It has nothing to do with a network drive, the issue is actually 2 physical drives. Within windows you can change the mountpoint of a physical drive to a folder that is under another physical drive. When I had it installed to the folder that was actually a different PHYSICAL volume it would error out reading in the icon paths and wouldn’t allow you to use the app. When you look at the error you can clearly see that the path is wrong due to what looks like a variable being used $VOLUME. 

 

If you want to setup a time I can work with you on how to repeat the issue because I validated twice that if you use the above mapping the app crashes.

bubnikv, 2018-02-12:
> auth/Aii-W61R84hY52h_cov-1aEniK5r1eAFks5tUBFngaJpZM4R8VUc> . <

Frankly your use case is quite rare and if the error happens inside the
Perl part, we will likely not fix it quickly. We are working on rewriting
the remaining Perl parts into C++, and then your issue will get the
attention it needs.

Which icons are missing? From the screenshots I may judge which part of the
application fails. Thanks.

eddie-moser, 2018-02-12:
It is the PERL crashing IIRC. The icons at the top for saving changes to filaments/printers etc… are missing.

bubnikv, 2018-02-12:
> It is the PERL crashing IIRC. The icons at the top for saving changes to filaments/printers etc… are missing.

Thanks. I will keep this issue open for reference, so we can test it once we have the UI code ported from Perl to C++.

bubnikv, 2018-03-14:
Isn't your issue the same as https://github.com/prusa3d/Slic3r/issues/732 ?

lukasmatena, 2023-11-12:
This issue is apparently outdated for a long time. Closing.

closed 2023-11-12 by lukasmatena