prusa3d/PrusaSlicer #712: "Don't Support Bridges"

sarf2k4, 2018-02-08:
### Version
1.39.1-alpha

### Operating system type + version
win 10 64

### Behavior
Full support even though "Don't Support Bridges" were turned on, changing the bridge angle will not do any changes.

To produce:
1. change bridging angle to 45'
2. turn on support
3. Yes to turn on "detect bridging parameter"
4. leave "Don't support bridges" ticked/enabled

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Comparison

[image]
[image]

FidelCapo, 2018-02-13:
Please, send me your STL and config files.
(File -> Export Config)

sarf2k4, 2018-02-13:
This time, I reproduced the problem again, but left the "Don't support bridges" ticked, it was solved. But turning it off then on again, it behave like the second picture of my first post. I'm attaching the zip file

[New folder.zip](https://github.com/prusa3d/Slic3r/files/1719619/New.folder.zip)

FidelCapo, 2018-02-14:
Thanks. It generates full support even if is Don't Support Bridges were turned on, because width of top layers is not large enough. If I scale up you model than Supports is generated correctly.

bubnikv, 2018-02-14:
So we admit there is an issue with the "don't support bridges" feature, we
see that for this particular issue the supports should not be generated. If
we scale your model, then the "don't support bridges" feature works as
intended.

bubnikv, 2021-02-24:
Bridges are only bridges if they are supported at both sides. Your area is supported at one side only, thus supports are correctly generated.

Your example is very much the same as provided for #261.

closed 2021-02-24 by bubnikv