prusa3d/PrusaSlicer #713: [Request] Brim Priority/Support on Brim

sarf2k4, 2018-02-08:
I would like to request this feature where the support will be printed on top of brims.

Current issues:
The object need to have support and brim enabled at once while some area will peel off without brim support.  Turning on overhang support will render brim not generated on that critical area that needs brim. Refer to the image below; the corner of the object part is are left out to give support's base priority over brim and during printing (with abs), it may peeled off by itself due to not enough adhesion.

Proposal:
Since the Support's base layer is also a single layer of plastic about 0.2mm, brim also the same about 0.2mm. The supports could just be printed on top of the brim layers.

[image]

FidelCapo, 2018-02-13:
Try download actual Slic3r verzion first. If that don't help, please, send me exported config and your STL file.

sarf2k4, 2018-02-13:
Hi, I've downloaded the original slic3r but there were no  exe file for windows (downloaded the zip from github). What I did was to copy the old slic3r executable for windows and copy on top of the latest zip file (I doubt the updated zip file would do anything).

The brim also behave the same thing where, it enclose the support base instead of overriding the support's base as brim. Though this might not be critical to those who're printing with PLA, but it is for ABS.

I attached a picture below generated using original slic3r 1.29 and between the support base and object's perimeter, there's a small gap about 0.4-1.0mm gap while the brim sticks with the support's base leaving the actual to be used object didn't get "brimmed" to let it stick to the bed.

P.S: I don't plan to use original slic3r for my everyday slicing program. 
[New folder.zip](https://github.com/prusa3d/Slic3r/files/1719661/New.folder.zip)
[image]

FidelCapo, 2018-02-14:
Thank you for report this Isuue.

ChrisRiddell, 2020-02-20:
[image]

I have this same problem I agree with @yoyo42 The brim should always be around the object then around the supports.

gerdmuller, 2020-03-07:
Have the same problem, so I was searching for a solution and found this request.
It's totally clear that the brim should have priority over the support pattern on the first layer. In succeeding layers, the supports should be based on the brim.

adamdport, 2020-03-11:
It looks like this is duplicated by #1156. However, [someone made a commit there a year and a half ago](https://github.com/prusa3d/PrusaSlicer/issues/1156#issuecomment-424767306) to resolve it. Is that something that can be merged?

bfeaginjr, 2020-12-18:
same issue, please fix this, as parts with curves on the Z plane have a hard time adhering if they occur from a thin part; If the brim could be directly underneath the supports, it would resolve this issue. Going to try Cura.

bubnikv, 2021-02-22:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1156
Closing this one

closed 2021-02-22 by bubnikv

bubnikv, 2021-02-23:
Implemented with 055d232