prusa3d/PrusaSlicer #714: [Request] Estimated print time in gcode file name

link0007, 2018-02-09:
I love the print time estimation in the new alpha. What would make it even better is if the print time could be incorporated in the filename. 

(What would make it downright amazing, is if the print time could be somehow communicated to the printer, and shown on the display during printing.)

bubnikv, 2018-02-12:
>I love the print time estimation in the new alpha.

Thanks.

> What would make it even better is if the print time could be incorporated in the filename.

We thought about it as well, but I am not sure whether it will make into the 1.39.1 release.

> (What would make it downright amazing, is if the print time could be somehow communicated to the printer, and shown on the display during printing.)

That is in the planning as well.

sarf2k4, 2018-05-23:
I've thought of having a new gcode command that takes in several parameters either by full seconds or by H,M,S for that gcode that hasn't been used yet for example m590 3600 (1H). Not just that this gcode also will prompt the user if the user really want to print the file rather than accidentally print something and turns out to be its a very long print.

bubnikv, 2018-12-12:
The Slic3r PE 1.42 alpha, which will be released soon, supports now the following additional placeholders in the output file template:

    "print_time", "normal_print_time", "silent_print_time", 
    "used_filament", "extruded_volume", "total_cost", "total_weight", 
    "total_wipe_tower_cost", "total_wipe_tower_filament"

The time is in the dhm format (5d3h1m for 5 days, 3 hours and 1 minute), rounded to a full minute.

Closing.

closed 2018-12-12 by bubnikv

einencool, 2019-02-24:
Hello, is there a way to get the Information for the "print_time" with an M117 Command in the start Gcode?
I don't figure out, what there will be the right placeholder...

Thank you in advance