prusa3d/PrusaSlicer #715: Estimated Print Time Missing

junkieball, 2018-02-10:
### Version
1.39.1 alpha 

### Operating system type + version
Mas OS High Sierra 10.13.3

### Behavior

I am probably missing a setting here.  I have installed a the latest alpha for the print time estimates.  I have a MK2s i can see the additional g-code in my start g-code, i have sliced a model but cannot for the life of me find where the estimate print time is located.  A little pointer in the right direction please?

CapnBry, 2018-02-10:
Have you saved the gcode? The estimate appears on the Platter tab on the bottom right Sliced Info box after the gcode file is either saved or sent to OctoPrint. It doesn't do it when you hit the slice button for some reason, same as it has with the filament usage estimate.

junkieball, 2018-02-10:
That's what was wrong.  I thought the time estimate would appear once sliced like cura.  Thanks for that.  Not sure it's very accurate though as just printing a piece and the time estimate is 26+ hours where as online using code.ws its just over 3 hours which i'm more inclined to believe as print has been running for just over an hour and is just under a third done.
[image]

bubnikv, 2018-02-12:
You likely have your slic3r wrongly configured. What firmware are you running? Where did you get your printer profiles from? Please read the release notes for the 1.39.1-alpha for explanation what the Slic3r needs for the time estimation to work correctly.

Blizzke, 2018-02-17:
Very nice feature that I've been missing from Cura fo quite a while!
However the way this is handled is not very user friendly.
There should be at either a button to request the timing/material usage without (having to write it to disk) , or there should be an "Export again" button that overwrites the previous file. If you're experimenting with settings it's just way too much of a hassle to have to go through that file dialog every time.

th33xitus, 2018-02-18:
@Blizzke Im pretty sure something like this will be implemented in the future. For now it is the first try of giving a precise print time estimation. And i really like it, since now Slic3r PE is the most accurate one. Even though it is not user friendly. I totally agree with you. In my cases the estimated print time differs only by a few minutes which is awesome compared to S3D's estimation (most of the time its not even close at all during multi-hour prints) or some webtools.

It would be super awesome to be able to set those settings in your printer settings tab in a new menu other than the custom g-code entry. Could be called "Kinematics" or something and then you write down your printer's acceleration, speed and jerk settings and whatever else needed.

atxkev, 2018-02-25:
Maybe there is a reason why they added it when you save but that wasn't well thought out.

Mike4U, 2018-02-28:
Seemed fairly close for me on one part, 4hr 20 min eat, 4hr 21 actual
Not on other 27.6m 8hr 10min est, 22m. 8r 10min actual

bubnikv, 2018-02-28:
@Mike4U 
> Not on other 27.6meter 8hr 10min est, 22meter 8r 10min actual

Please provide us with the data, so we can look into it.
The time is precise, not the filament length. Where did you get the actual filament length consumed?

bubnikv, 2018-02-28:
@atxkev 
> Maybe there is a reason why they added it when you save but that wasn't well thought out.

We are very well aware, that it is not quite optimal, that the user has to do the G-code export. This is due to the current Slic3r architecture, and we will be improving upon it in the following releases.

Mike4U, 2018-02-28:
@bubnikv 
> Please provide us with the data, so we can look into it.
The time is precise, not the filament length. Where did you get the actual filament length consumed?
> 
Please forget about my numbers.  The actuals were from the previous version and I can't reliably say that the setting were the same.  Sorry

bubnikv, 2018-05-10:
@Blizzke It would be super awesome to be able to set those settings in your printer settings tab in a new menu other than the custom g-code entry. Could be called "Kinematics" or something and then you write down your printer's acceleration, speed and jerk settings and whatever else needed.

It will likely come one day, because for the Prusa's printers we want to have two time estimates: One for the silent mode and one for the full power mode.

@Mike4U Thanks for the response. So it seems everything is working now. Closing.

closed 2018-05-10 by bubnikv