prusa3d/PrusaSlicer #716: Incorrect slicing

rashn, 2018-02-11:
### Version
1.39.1-alpha-prusa3d

### Operating system type + version
Mac OS High Sierra 10.13

### Behavior
Incorrect slicing. Screenshots and STL are attached.

#### STL/Config (.ZIP) where problem occurs
[coverka.stl.zip](https://github.com/prusa3d/Slic3r/files/1714293/coverka.stl.zip)
[image]
[image]

Sebastianv650, 2018-02-11:
Most likely because the stl has 46 errors stated by Slic3r. If you repair it, for example using Meshmixer, it should be OK.

bubnikv, 2018-02-12:
Based on feedback of @enricoturri1966, the following change https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01 broke the STL import for this model.

BirkBinnard, 2018-02-12:
The best source for fixing STL files that I've found is this one: https://service.netfabb.com/login.php. It works better than the 3D Builder app, better than the free version of NetFabb, and I run  all my STL files through it before any slicing.

bubnikv, 2018-02-12:
Interestingly, I believe that the 3D Builder application sends the STLs to
the same service, so it shall work the same way :-)

closed 2018-02-23 by bubnikv

BirkBinnard, 2018-02-23:
Nice job on closing this one - does it mean we could eliminate the problem by running a "check normals" routine on all geometry before exporting as an STL file? Or does it mean that 3D design programs can't be trusted to get normals correct in all cases, no matter what is done at design time?

Sebastianv650, 2018-02-23:
> Or does it mean that 3D design programs can't be trusted to get normals correct in all cases, no matter what is done at design time?

Given my experience, it only depends on the design program. I never got a single mesh error on my models using Catia or Fusion 360. Therefore I would say it's basicaly a bug in design tools..

BirkBinnard, 2018-02-23:
Well, you could be correct of course. Catia is not anything I have access to, but since it is the top of the heap of all 3D design software I am not surprised it works well. I did try Fusion 360, but it just seemed so counter-intuitive to me I gave up on it. 

I use a combination or Rhino and Grasshopper, and I have known for a long time that the STL files Rhino produces can (and sometimes do) have errors. I did try various methods to fix these (eliminating naked edges, merging meshes or Rhino's internal objects called BReps, checking normals, etc.), but doing that was tedious and not always helpful. So I reverted to just running my STL files through the Netfabb fixer as part of my overall workflow.

bubnikv, 2018-05-10:
@FidelCapo Would you please test this model, and if it slices correctly in the upcoming 1.40 alpha, would you please add ti to the list of test objects? Thanks.

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.

commit 1489b990 (2018-02-15): Added workaround into admesh stl_fix_normal_directions() function to prevent meshes to be broken by the repairing process (fixes #716, #574, #413, #269, #262, #259, #230, #228, #206)