prusa3d/PrusaSlicer #717: Save project / factory file - feature request

hoegge, 2018-02-13:
### Feature request
It would be great if you could save the plate and all active settings in a single "factory" file, so you can transfer everything and save for future retrieval later even if any configurations has been changed. This should also include settings that are not saved in a print, filament or printer configuration.

### Explanation / details
I have just "discovered" Slic3r after having used Simplify3D for a while. Slic3r is in many ways much more intuitive especially how the settings are organised. One thing I really miss is the ability to save a "factory" file that contains everything for a specific print. It saves all settings, STLs and plate arrangements, so you know you can accurately do the exact same print again at some time later without having to spend a lot of time checking what are the right settings. Especially when you have had to tweak and change settings specifically for a model to make it just right, then you don't want to save a separate config for every print to be able to reproduce it and then a "factory file" would be very useful.

bubnikv, 2018-02-13:
The "factory" file is currently already nearly there, you just need to export the platter as an AMF file and also export the config. The two files together make an equivalent of the Simplify3D factory file.

@enricoturri1966 is currently working on integrating the config file into the AMF file. He also works on support of the 3MF format, which will also be able to integrate the config file, making the AMF and 3MF formats the complete "factory file" equivalent.

hoegge, 2018-02-13:
Thanks @bubnikv - I know the workaround with the plate and ini file - works ok. Would be great with one file - but understand you are already working on that. Great.

bubnikv, 2018-03-14:
This should be working with the current master, and it will be part of the upcoming 1.40 release. The 3mf format is supported, and the complete config.ini is included into both the 3mf and amf file formats. Closing.

closed 2018-03-14 by bubnikv