prusa3d/PrusaSlicer #718: Wipe tower disappears when 'Single Extruder Multi Material' is deselected in Printer Settings.

joris57, 2018-02-13:
### Version
1.38.6

### Operating system type + version
Windows 10

### Behavior
When 'Single Extruder Multi Material' is deselected in Printer settings, the wipe tower disappears, even though 'Wipe tower' is selected in Print Settings

bubnikv, 2018-02-14:
this is intentional, the wipe tower currently only works in the single
extruder multi material mode.

joris57, 2018-02-14:
Thanks for the quick reply.

Am I right that this means that the wipe tower can only be used with an Original Prusa i3 Mk2(s)/Mk3 with MMU and not with any other multi material printer (since the load/unload for the MMU is also generated when switching on 'Single Extruder Multi Material'?

bubnikv, 2018-02-14:
Yes. Due to the limited resources  of our development team, the wipe tower
is limited to the  Original Prusa i3 Mk2(s)/Mk3 with MMU for now. It will
work with other single material mutli extruder setups, but currently there
are some values like the bowden length hard coded. @lukasmatena is working
on reparametrization of the wipe tower, so we are making the wipe tower
more generic, but I am not sure when we will support the multi extruder
setups.

joris57, 2018-02-23:
That's disappointing, but I did find a workaround on how to create my own wipe towers and print them at the right time. Looking forward to have smart wipe towers available for non-Prusa-MMU multi-material printers.

Closing the issue for now.

magkopian, 2018-02-23:
@joris57 Can you share your workaround please, I'm really interested in the ability of printing wipe towers with a dual hotend as well.

joris57, 2018-02-24:
As a workaround, I'm creating two or more (depending on the number of extruders used) pillars on the printbed that are assigned to the different extruders, I want to print the model with.

Within each layer, these pillars are printed first, when starting to print their respective colors (making sure that the transition material of an extruder change is deposited there and not in the main model).

In my case, the pillars are 10x10mm with an infill of 50%. That is plenty extrusion for a color change with the original Diamond hotend, I am using. 

There are two issues, when doing this:
1) The infill of the pillars needs to be set independently of the infill of the main object. If not, you may find yourself wasting material or depositing insufficient material.
2) The sequence of printing needs to be guaranteed, so that the priming pillars are always printed first.  

I'm using two Slic3r features for this:
1) The ability to assign individual settings to an object.
2) The fact that objects are printed in the sequence they are dropped on the build plate (for each extruder).

So the actual wokflow (for my 3 extruder printer, to be adjusted for more or less extruders):
1) Drop 3 10x10x10mm cubes on the build plate.
2) Assign extruders 1-3 to the respective cubes and set infill to 50% (Right-click on the object in the Plater, then select Setings. Extruder field is readily available in Object Settings, Fill Density can be aded by clicking the green +-sign).
3) Drop the main object(s) on the build plate and assign extruders.
4) Scale the pillars to the height of the object (Right-click on the pillars > Scale to size > Along Z axis).

Slice and print.

You can save the build plate to an .amf file, so that you don't have to repeat steps 1 and 2 (Plater > Export plate as AMF).

Hope his is of use to you.

magkopian, 2018-02-25:
@joris57 Thank you for talking the time to explain your approach in detail. I'll try your method and see how it goes.

joris57, 2018-05-10:
Hi Lukas,

This sounds very promising.

As far as 'multi-nozzle multi material' is concerned (which I assume is meant when you refer to 'multiple extruder'), I have following comments:   

> - There is no need for the ramming phase before the toolchange. ==> Agree
> - There is no need to cool the filament after unloading. ==> Disagree. Cooling down the inactive hot-end may reduce oozing (but will increase material change time).
> - The still IS the need to prime the "new" filament before it's used. ==> Only at the start of the print (priming line), followed by a retraction. **At the first use, a standard 'unretract' should be performed.**

The 'cooling down' parameter is in standard Slic3r stored in the printing settings (ooze control).

What would be beneficial is a more intelligent ooze shield, but I assume that is outside the scope of this topic?

lukasmatena, 2019-08-26:
The wipe tower can now be used with multi extruder multimaterial printers in PrusaSlicer 2.1.0.
Closing.

closed 2019-08-26 by lukasmatena