prusa3d/PrusaSlicer #720: Building With Visual Studio 2017

jonnyhsu, 2018-02-14:
### Version
version_1.40.0-alpha-1-g6e80a91

### Operating system type + version
Windows 10

### Behavior
I'm trying to build this on Windows with VS 2017 and Strawberry Perl. Has this been done before? I'm running into quite a bit of trouble. Here's where I am so far:

1. `perl Build.PL` runs ok
2. `perl Build.PL --gui` fails because Wx won't install. After digging through the logs, it turns out that it's Alien::wxWidgets trying to build wxwidgets, but the prdownloads.sourceforge.net site is returning a 403 (down?). So I manually download wxwidgets 3.0.2 source, and populate the Alien directory manually, which it auto detects when I manually build Alien:wxWidgets (http://search.cpan.org/dist/Alien-wxWidgets/README.txt). This allows Wx to install.
3. `cmake .. -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release`
This fails because it can't find libperl. Not sure why, but one solution is to spoon feed it the libperl library: `cmake .. -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release -DPERL_LIBRARY=C:/Strawberry/perl/lib/CORE/libperl526.a`

Now it can't find TBB (I have it in the recommended c:/local/tbb). Solution: update to latest `FindTBB.cmake` file (https://github.com/justusc/FindTBB) and feed it the cmake directory. So now the command looks like this: `cmake .. -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release -DPERL_LIBRARY=C:/Strawberry/perl/lib/CORE/libperl526.a -DTBB_ROOT_DIR=c:/local/tbb`

Now that works.

4. `nmake` fails with dozens of errors (warning C4244: 'initializing': conversion from 'double' to 'coord_t', possible loss of data). I am running the TBB 2018 Update 2, and boost 1.63 for lib64-msvc-14.0. The final nmake error:
`NMAKE : fatal error U1077: 'C:\PROGRA~2\MIB055~1\2017\COMMUN~1\VC\Tools\MSVC\1411~1.255\bin\Hostx64\x64\cl.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.11.25503\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.11.25503\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.`

All the Visual Studio 2013 download pages forward to 2017, so it doesn't seem like 2013 is available anymore. Is building with 2017 possible?

supermerill, 2018-02-14:
> All the Visual Studio 2013 download pages forward to 2017

try
https://www.visualstudio.com/vs/older-downloads/

bubnikv, 2018-02-14:
Thanks, I have updated the wiki with that "older downloads" link.
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

bubnikv, 2018-02-14:
> Slic3r-Prusa-Edition-on-MS-Windows

> Is building with 2017 possible?

Likely it is, but you would have to compile the whole perl and all the perl
libraries yourself, which is a lenghty, manual process, which I don't
recommend to waste time with.

I hope we will get rid of the Perl dependencies before we will be forced to
depart from the Visual Studio 2013.

dartrax, 2018-02-16:
Hi,
what about Visual Studio 2015? I have that already installed and I wonder if I need to install VS 2013 too.

I needed to download and install cmake manually, is that correct (and may be missing in the wiki)?

Now I'm stuck at the `cmake .. -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release
nmake` command. It gives me:

```
--  PerlEmbed_LDDLFLAGS    = -dll;-nologo;-nodefaultlib;-debug;-opt:ref,icf;-ltcg;-libpath:c:\wperl64\lib\CORE;-machine:AMD64;-subsystem:console,5.02
CMake Error at C:/Program Files/CMake/share/cmake-3.10/Modules/FindBoost.cmake:1928 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.63.0

  Boost include path: C:/local/boost_1_63_0

  Could not find the following static Boost libraries:

          boost_system
          boost_filesystem
          boost_thread
          boost_log
          boost_locale
          boost_regex

  Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  xs/CMakeLists.txt:474 (find_package)

--   TBB_INCLUDE_DIRS        = C:/local/tbb/include
--   TBB_DEFINITIONS         =
--   TBB_LIBRARIES           = C:/local/tbb/lib/ia32/vc12/tbb_static.lib
--   TBB_DEFINITIONS_DEBUG   = -DTBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = C:/local/tbb/lib/ia32/vc12/tbb_static_debug.lib
--   TBB_DEFINITIONS_RELEASE =
--   TBB_LIBRARIES_RELEASE   = C:/local/tbb/lib/ia32/vc12/tbb_static.lib
--   AlienWx_FIND_COMPONENTS=base;core;adv
compilet-Tyl0j.c
   Bibliothek "compilet.lib" und Objekt "compilet.exp" werden erstellt.
msvcrt.lib(dll_dllmain.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_initterm".
msvcrt.lib(dll_dllmain.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_initterm_e".
msvcrt.lib(dll_dllmain.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "__telemetry_main_invoke_trigger".
msvcrt.lib(dll_dllmain.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "__telemetry_main_return_trigger".
msvcrt.lib(dll_dllmain.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "__C_specific_handler".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "__C_specific_handler".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_seh_filter_dll".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_initialize_narrow_environment".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_initialize_onexit_table".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_register_onexit_function".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_execute_onexit_table".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_crt_atexit".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_crt_at_quick_exit".
msvcrt.lib(utility.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "_cexit".
msvcrt.lib(tncleanup.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "__std_type_info_destroy_list".
msvcrt.lib(utility_desktop.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "terminate".
msvcrt.lib(utility_desktop.obj) : error LNK2001: Nicht aufgel�stes externes Symbol "memset".
compilet.dll : fatal error LNK1120: 16 nicht aufgel�ste Externe
--   AlienWx_VERSION           = 3.000003
--   AlienWx_INCLUDE_DIRS      = C:\wperl64\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib;C:\wperl64\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\include
--   AlienWx_LIBRARIES         = -LIBPATH:C:\wperl64\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib;wxmsw30u_core.lib;wxbase30u.lib;msvcprt.lib;wxbase30u.lib;wxmsw30u_core.lib;wxmsw30u_adv.lib
--   AlienWx_LIBRARY_DIRS      =
--   AlienWx_DEFINITIONS       = -DWIN32;-D_CRT_SECURE_NO_DEPRECATE=1;-D_CRT_NON_CONFORMING_SWPRINTFS=1;-D_SCL_SECURE_NO_WARNINGS=1;-D__NO_VC_CRTDBG__;-D__WXMSW__;-DNDEBUG;-D_UNICODE;-DWXUSINGDLL;-D_WINDOWS;-DNOPCH
--   AlienWx_DEFINITIONS_DEBUG =
--   AlienWx_CXX_FLAGS         = -GF;-TP;/c;/nologo;/TP;/MD;/Zi;/O2;/W4;/GR;/EHsc
-- Configuring incomplete, errors occurred!
See also "C:/local/Slic3r/build/CMakeFiles/CMakeOutput.log".
See also "C:/local/Slic3r/build/CMakeFiles/CMakeError.log".
```

Some suggestions? Otherwise I'll check if I can find some space on my HD to install VS2013 tomorrow.

supermerill, 2018-02-16:
> I needed to download and install cmake manually, is that correct (and may be missing in the wiki)?

yes, latest version was ok for me.
for the boost problem, it shouldn't appear is it's installed in c:/local. Add them to your master cmakelist if it can't find them (like for tbb).

dartrax, 2018-02-17:
Hi, thanks for the suggestions.
In fact, boost is installed in c:/local, but while scratching my head why it couldn't be found it became clear to me that I need the boost for msvc 14 version since I still tried with VS 2015. After that, I got a presumably similar error with `nmake` like you:
```
[ 98%] Linking CXX shared library XS.dll
LINK: command "C:\PROGRA~2\MICROS~1.0\VC\bin\amd64\link.exe /nologo @CMakeFiles\XS.dir\objects1.rsp /out:XS.dll /implib:XS.lib /pdb:C:\local\Slic3r\build\xs\XS.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO libslic3r.lib libslic3r_gui.lib admesh.lib miniz.lib clipper.lib nowide.lib polypartition.lib poly2tri.lib OpenGL32.lib GlU32.lib C:\wperl64\lib\CORE\perl524.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_system-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_filesystem-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_thread-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_log-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_locale-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_regex-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_chrono-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_date_time-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_atomic-vc140-mt-1_63.lib C:\local\boost_1_63_0\lib64-msvc-14.0\libboost_log_setup-vc140-mt-1_63.lib C:\local\tbb\lib\ia32\vc12\tbb_static.lib -LIBPATH:C:\wperl64\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib wxmsw30u_core.lib wxbase30u.lib msvcprt.lib wxbase30u.lib wxmsw30u_core.lib wxmsw30u_adv.lib expat.lib glew.lib msvcprt.lib wxmsw30u_adv.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:XS.dll.manifest" failed (exit code 1112) with the following output:
XS.cpp.obj : Es wurde ein mit /GL kompiliertes MSIL .netmodule oder module gefunden. Verkn³pfung wird mit /LTCG erneut gestartet. F³gen Sie /LTCG zur Linkbefehlszeile hinzu, um die Linkerleistung zu erh÷hen.
tbb_static.lib(tbb_misc.obj) : fatal error LNK1112: **Modul-Computertyp "X86" steht in Konflikt mit dem Zielcomputertyp "x64".**
NMAKE : fatal error U1077: ""C:\Program Files\CMake\bin\cmake.exe"": Rückgabe-Code "0xffffffff"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"": Rückgabe-Code "0x2"
Stop.
```
Ok, I'll try with VS2013 now ;-)

dartrax, 2018-02-17:
With VS2013 I've got the same error. I've found out that the reason was #679, I was referencing the wrong TBB_LIBRARY (C:/local/tbb/lib/ia32/vc12 instead of C:/local/tbb/lib/intel64/vc12)
Fixing that, deleting caches and rebuilding everything `nmake `completes now. Now I do not know if it would have worked with VS2015...

Now i'm stuck with this:
```
c:\local\Slic3r>perl slic3r.pl
Can't locate object method "new" via package "Slic3r::GUI::BedShapePanel" (perhaps you forgot to load "Slic3r::GUI::BedShapePanel"?) at C:/local/Slic3r/lib/Slic3r/GUI/ConfigWizard.pm line 376.
```

> nmake fails with dozens of errors (warning C4244: 'initializing': conversion from 'double' to 'coord_t', possible loss of data). 

I've got them too, with VS2015 and VS2013 respectively. Is that normal?

jonnyhsu, 2018-02-20:
Not sure if the C4244 warning's are "normal" or not, but I did eventually get everything built and running with VS2013 and still got those warnings.

supermerill, 2018-02-21:
There are some pull request to remove warnings. Ignore them for now.

bubnikv, 2018-02-21:
Regarding the missing environment variables in the perl starter batch file,
we will release updated perl packages.

@VojtechKral is working on integrating SSL networking into the C++ side, so
we will add another binary dependency likely the same way we added the TBB.

> Now I do not know if it would have worked with VS2015...

Most likley not. Visual studio and likely most C++ compilers are known to
have incompatible C++ ABI between major versions. We provided binary builds
for the perl environment including the wxWidgets and the wxWidgets/Perl
bindings, and we will use this distro dependent on VS2013 until we are
forced to upgrade the visual studio, or better until we get rid of the Perl
completely.

So sorry, we will not support newer visual studios in the near term, as
long as we don't get a helping hand outside of Prusa Research.

If you are willing to help, there is a looong manual sequence documented of
how I compiled our Windows perl binary distro. You are most welcome to help
out and do that for VS2015 or VS2017, but be warned that it took me likely
a full working week to come up with the first build, and likely couple of
working days more to produce the binary perl zips, which all of us are
using.

https://github.com/prusa3d/Slic3r/wiki/Building-Perl-&-wxWidgets-with-Visual-Studio-2013

To support testing of your development builds, we may run our build server
against your fork on request. That way you will get a build of the same
format / content / size as our official builds, and they will be marked
with your github user name and your branch to be clearly differentiated.
This would require a bit of additional work from our side on the build
server and on the infrastructure, and we may need to beef up our build
server HW.

jonnyhsu, 2018-02-21:
I was able to build perl and got maybe halfway through the supporting packages with VS2017 before deciding it wasn't worth the time right now. Perl itself actually compiled perfectly on the first try with nmake, so it's the packages that cause all the trouble.

> @VojtechKral is working on integrating SSL networking into the C++ side, so we will add another binary dependency likely the same way we added the TBB.

Isn't there an SSL library in boost?

vojtechkral, 2018-02-22:
> Isn't there an SSL library in boost?

Yes, but it's only a fairly thin wrapper on top of OpenSSL. I'm using libcurl instead (which also requires OpenSSL, but only no Linux et al., not on Windows and OS X, where it is able to use the native SSL implementation.)

N4rki, 2018-03-05:
So I followed the instructions and thanks to the people here I it worked for the most part.
The only thing that failed is `ctest --verbose`. It can't find two files/directories although both do exist. I started cmdline as admin so it can't be an issue with access rights.

<details> <summary> click here for the cmdline output </summary>

```
C:\local\Slic3r\build>ctest --verbose
UpdateCTestConfiguration  from :C:/local/Slic3r/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :C:/local/Slic3r/build/DartConfiguration.tcl
Test project C:/local/Slic3r/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: xs

1: Test command: C:\wperl64d\bin\perl.exe "C:/wperl64d/bin/prove" "-I" "C:/local/Slic3r/local-lib/lib/perl5"
1: Test timeout computed to be: 9.99988e+06
1: Can't open perl script "C:/wperl64d/bin/prove": No such file or directory
1/2 Test #1: xs ...............................***Failed    0.05 sec
test 2
    Start 2: integration

2: Test command: C:\wperl64d\bin\perl.exe "C:/wperl64d/bin/prove"
2: Test timeout computed to be: 9.99988e+06
2: Can't open perl script "C:/wperl64d/bin/prove": No such file or directory
2/2 Test #2: integration ......................***Failed    0.02 sec

0% tests passed, 2 tests failed out of 2

Total Test time (real) =   0.09 sec

The following tests FAILED:
          1 - xs (Failed)
          2 - integration (Failed)
Errors while running CTest

```
</details>

But `perl` slic3r.pl` works fine. Is ctests failing a problem none theless?

What I'm really wondering though is how exactly I now load this into Visual Studio 13. I couldn't find any project files that can be opened with Visual Studio. The whole purpose of this is to use the IDE and it's debugger right? I tried several methods like importing the Slic3r folder into VS as a new project but If I doing so compiling within the IDE fails.
I know that cmake can generate the project files via `cmake .. -G "Visual Studio 12 2013" and it does so rather swiftly and with no errors. But same issue here: Building within Visual Studio 2013 fails unfortunately.

So what's the right way or what am I missing?

vojtechkral, 2018-03-05:
@N4rki Regarding `ctest`, I'm not sure, I'll look into it / ask around...

Regarding Visual Studio project files, I am currently working on simplifying the build process, this is done in the [http+build](https://github.com/prusa3d/Slic3r/tree/http%2Bbuild) branch. I have written [a preliminary how-to](https://github.com/prusa3d/Slic3r/blob/http%2Bbuild/doc/How%20to%20build%20-%20Windows.md) which is meant as an upgrade of the existing how-to. The branch contains some improvements that should provide Visual Studio debugging out of the box, if all goes well. This is described in the how-to.

It is sill in a somewhat work-in-progress state, but I've getting good results on Windows and you can give it a try if you want. I'd certainly appreciate outside feedback.

N4rki, 2018-03-05:
@vojtechkral Thank you for that quick response. I've been working on it for  a while now trying out different things. So I got to work with my initial wperl64d + local folder setup taken from [here](https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows#compile-using-a-free-visual-studio-community-2013). What made all the difference was simply taking your command `cmake .. -G "Visual Studio 12 Win64" -DCMAKE_CONFIGURATION_TYPES=Release;RelWithDebInfo` instead of what I had used earlier. Here I still had to add `-DTBB_ROOT_DIR=c:/local/tbb`, which is one of the things you fixed in your new how-to.
Also making the project "Slic3r" the startup project is crucial, otherwise I get an "access denied" error. Although it might seem obvious to most people this is really something were a newbie could get stuck very easily so details like this are important to include.

So next up I followed your how-to and it worked fine. 

So the easiest way to pass parameters to the main function of slic3r is to go right click on "slic3r" -> properties -> config properties -> Debugging -> insert desired commands in "Command Arguments" line. Is this correct? I had to look that up first so maybe it can be included as well.

Now is where the confusion starts:

If I pass the arguments "torus.stl" "--export-svg" "--layer-height 0.2" the program runs until the end but not a lot happens. There are a few PDB files that are not found. I don't know if that's an issue.
<details> <summary>
Here's the command box output: </summary>

```
'slic3r.exe' (Win32): Loaded 'C:\Users\VB\Documents\Bachelorarbeit\Code\Slic3r_prusa3d\Slic3r\build\xs\RelWithDebInfo\slic3r.exe'. Symbols loaded.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\msvcp120.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\msvcr120.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\powrprof.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\profapi.dll'. Cannot find or open the PDB file.
'slic3r.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. Cannot find or open the PDB file.
The thread 0x2cb0 has exited with code 0 (0x0).
The thread 0x166c has exited with code 0 (0x0).
The thread 0x34c4 has exited with code 0 (0x0).
The program '[10152] slic3r.exe' has exited with code 0 (0x0).
```

</details>
<br>
Only then I realized that most of the code in slic3r.cpp is disabled through a `#if 0 (...) #endif` clause.

When comment out these two statements and try to build. I get 122 errors.

Is there a way to make all of it compile? Why is most of the code in slic3r.cpp commented out anyway?

And lasty, I assumed that the whole program can be debugged via Visual Studio, so including the gui and being able to interactively use slic3r while running it from the IDE. This seems not to be the case, or am I missing something? So the best case scenario is to get the CLI only version running in Visual Studio?

Anyway you already made it so much easier to get to where I am at right now so big up for that!

vojtechkral, 2018-03-07:
@N4rki Thanks for testing this! As you probably already know my changes are merged in the master now. Anyway, those missing PDB files are not an issue, that's just some system libraries.

As for running the slic3r, it is always run through perl. There is no standalone C++ main function (at least not yet), and so you run the slic3r as `perl slic3r.pl --foo --bar` or just `perl slic3r.pl` to get the GUI. All C++ code is invoked from Perl.

> And lasty, I assumed that the whole program can be debugged via Visual Studio, so including the gui and being able to interactively use slic3r while running it from the IDE.

Yes, that's idea. 

> The reason of my inquiry is to get this working with alexrj's slic3r.

The upstream slic3r doesn't use CMake, so probably most of our how-to is not applicable.

N4rki, 2018-03-07:
Okay thanks for clearing all that up :)

I'm starting to understand the whole concept now. We have to attach Visual Studio to the created perl - slic3r process in order for it to work right? So Debug -> Attach to process -> select the perl interpreter.

This way it works, my breakpoints are getting hit etc. So you might want add that detail to the guide :)

vojtechkral, 2018-03-07:
@N4rki Attaching may work as well, but I aim for that to not be necessary, MSVC should be able to launch slic3r through the Perl process. However there was a bit of a bug with my MSVC setup which I'm fixing now and I'll publish the changes in a bit...

vojtechkral, 2018-03-12:
@N4rki The fixes are now integrated, debugging should now hopefully be smoother...

bubnikv, 2018-03-12:
> Only then I realized that most of the code in slic3r.cpp is disabled through a `#if 0 (...) #endif` clause. When comment out these two statements and try to build. I get 122 errors.
> 
> Is there a way to make all of it compile? Why is most of the code in slic3r.cpp disabled anyway?

The slic3r.cpp is there for the future, to be a base for the command line only slic3r. As of now, this piece of code is unused in our fork. 

As you likely already found out, our fork has to be started through the perl interpreter.

bubnikv, 2018-03-14:
@vojtechkral updated the CMake scripts. Does this work for you?
https://github.com/prusa3d/Slic3r/blob/master/doc/How%20to%20build%20-%20Windows.md

N4rki, 2018-04-19:
Hi bubnikv, sorry for not getting back; that message whether it worked now for me somehow slipped through the cracks. I just cloned the newest master and followed the instructions in the guide.
Building via Visual Studio work fine but when I run it, I get an error and perl exits with code 255 (0xff). It says: 
```
Loading an application config file failed:
<unspecified file>(6): '=' character not found in line
```
I get the same error though when I try to run via `perl slic3r.pl`. 

I'm not in need to build Slic3r PE anymore, but this might be helpful to you. Its probably something minor as I had it working by 'attach to process' previously. 
So assuming this normally works,  you did tremendous work here because providing a good and easy way to get it working with Visual Studio really helps getting new people to the project.

As I had explained my main focus lies in the upstream Slic3r because of the 3mf capabilities as well as slicing to SVG. But to this day I couldn't get any good setup working to debug it, even GDB is no good because of perl. So I'm using an editor to code, compile the whole thing after each change and use CLI printouts to know what's happening. This made it really tough for me, especially since I haven't done anything with perl or c++ before. Would be so good if what you did was somehow transferable to the upstream Slic3r but I think there is quite some work and restructuring needed right? 
Anyway, keep up the good work!

- Val

PetrGasparik, 2019-05-09:
VS2017 compilation (under VS2019 dev console) ends in failure to find BOOST libraries:
```
  Unable to find the requested Boost libraries.

  Boost version: 1.63.0

  Boost include path: C:/local/include/boost-1_63

  Could not find the following static Boost libraries:

          boost_system
          boost_filesystem
          boost_thread
          boost_log
          boost_locale
          boost_regex
```
Enviroment:
- Static libraries are in C:\local
- set: 
```
BOOST_LIBRARYDIR="C:\local\lib"
BOOST_ROOT="c:\local\include\boost-1_63\boost"
```
- command:
`cmake .. -DCMAKE_PREFIX_PATH="c:\local"`

supermerill, 2019-05-11:
@PetrGasparik i manually set these on cmake-gui, from the deps build directory (deps built with vs2017)

It works.