prusa3d/PrusaSlicer #724: [Feature Request ] Allow the sacrificial skirt have a brim

magkopian, 2018-02-18:
### Current Behavior
Currently, even if the `Distance from the object` setting of the skirt is set to 0, there is still a gap between the brim and the skirt. This makes it impossible to have a sacrificial skirt with a brim, which results to poor adhesion with the build plate due to the small contact surface.

Of course it is possible to enhance the adhesion to the build plate by just increasing the minimum loop count, but especially with larger prints even a few extra loops can contribute a lot to the material waste.

### Proposed Solution
In my opinion, the best solution would be to have an additional setting for specifying the loop count just for the first layer of the sacrificial skirt. With that set to a larger number than the regular loop count, it would effectively result to a  sacrificial skirt that has a brim.

If that is not possible, another solution would be to simply allow the skirt to touch the regular brim when the `Distance from the object` is set to 0. With the later the skirt is not going to have a brim from both sides, but it's still better than nothing.

bubnikv, 2018-05-10:
By the way, what do you use this high skirt for?

magkopian, 2018-05-11:
@bubnikv The sacrificial skirt? It's for catching the oozing from the non-printing nozzle while the print head moves around. The sacrificial skirt gets generated automatically when you check the `Enable` option under the `Ooze prevention` at the `Multiple Extruders` section. It's what other slicers call ooze shield.

bubnikv, 2018-05-11:
yeah, that is what I expected. You really miss the ooze shield in Slicer.

rongith, 2018-06-02:
Morevover, If you have some "Elephant foot", the skirt don't touch the brim.

lukasmatena, 2019-10-01:
The logic of brim/skirt generation was recently enhanced. Skirt now respects the requested distance to object (it is no longer pushed outwards by object brim). It can therefore intersect the brim. In case that happens, the brim extrusions are split and a gap for the skirt is inserted. This should help attaching the skirt to the bed:
[image]

The described behaviour will be in PrusaSlicer 2.2.0 release.

I believe this solves the issue, so I'm closing it.

closed 2019-10-01 by lukasmatena

Nerbo77, 2020-10-05:
Hi, skirt, draft shield, ooze prevention, all three make just one shield for all the printing parts, then when printing sequentially or when printing multiple items, this dosen't work, maybe is better if we can choose, skirt, draft shield or ooze prevention individually for every printing part, or for all the platform.