prusa3d/PrusaSlicer #725: Support material not generated after modifying "XY separation between object and its support" parameter

ruedli, 2018-02-18:
### Version
1.38.6

Settings for the Prusa i3 

MK3

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
windows 10, 64bit OS

### Behavior
* After a change (increase in my case) of the parameter "percentage XY separation between object and its support", and slicing, the preview shows no generated support, just a ring on the baseplate, although supports should have been generated.

After saving the parameters, exiting Slic3r and opening Slic3r again, supports are generated as expected, the modified and save parameter is properly taken into account.

_Is this a new feature request?_ No, it seems like a bug.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[XY separation between object and support parameter problem.zip](https://github.com/prusa3d/Slic3r/files/1734762/XY.separation.between.object.and.support.parameter.problem.zip)

ryanneufeld, 2018-02-28:
I just ran into this today. Once you change this setting, the UI refuses to show any support or waste tower in the preview screen.

ruedli, 2018-02-28:
Indeed, ran into it 3 more times; every time when I change this setting.

I for now save any modified settings, exit slic3r, open it again and then
it slices fine. I do not know whether it affects only the displaying, or
also the gcode, but I do not want to take the risk.

bubnikv, 2018-03-01:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/709
and it has been fixed with https://github.com/prusa3d/Slic3r/commit/6f92424bab82f1795697b21e074e4652529271ff
in https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta

Please use https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta1
which will be released as a final release any time now.

closed 2018-03-01 by bubnikv