prusa3d/PrusaSlicer #727: Allow settings modifiers on vase mode prints

JesusFreke, 2018-02-18:
### Version
1.39.1-beta-prusa3d-linux64

### Operating system type + version
Ubuntu 17.10

### Behavior
I need to change the print settings for only part of a vase-mode model. I tried using the "load generic" functionality in the "settings" dialog to load a slab and change the extrusion width for that slab. Afterwards, when I try to slice it, I get the following error: "Configuration is not valid: The Spiral Vase option can only be used when printing single material object."

My work around is to slice the model twice with different settings, and then manually copy over the gcode for the necessary layers from the 2nd slice.

_Is this a new feature request?_
Yes

loudaslife, 2018-04-28:
I ran into this same problem and was quite annoyed, so I managed to compile a version that removes this arbitrary restriction. In `xs/src/libslic3r/Print.cpp` I commented out the following lines:

    if (this->regions.size() > 1)
        return "The Spiral Vase option can only be used when printing single material objects.";

Compiling with these lines disabled fixes the issue and allows me to use modifiers on vase mode prints. I've printed a couple models like this now to confirm that the gcode output works as intended, at least for preventing bottom solid layers from generating in the middle of my parts. However, commenting out these lines most likely breaks the intended functionality of preventing vase mode from being used in multi-material prints. A few other specific settings modifiers would also interfere with vase mode, depending on where exactly they're applied, so I'm honestly not sure how to fix this issue.

Implementing #798 could potentially provide a more elegant workaround to the whole problem.

Henry3DP, 2020-06-09:
As @loudaslife found a quick workaround but I have no idea how to compile a software. Could you or any other user provide a already compiled version of PrusaSlicer with this workaround?

francisco-pereira, 2020-06-10:
@Henry3DP, I can build a Linux version if you like, but no promises that it will work on your system.

Henry3DP, 2020-06-10:
@francisco-pereira I am a MacOS User so I am a complete newbie when it comes to Linux, but I have a raspberry around. So it is at least worth a try!?

bubnikv, 2021-06-03:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/4984
https://github.com/prusa3d/PrusaSlicer/issues/4230
https://github.com/prusa3d/PrusaSlicer/issues/727

AdeAnima, 2023-08-07:
Crazy how this is still a problem after 5 years 😅

dmytro-dov, 2025-06-14:
> Crazy how this is still a problem after 5 years 😅

5? Try 7.

closed 2026-09-29 by prusaslicer-bot[bot]