prusa3d/PrusaSlicer #729: Print profiles wrong??

chilternflyer, 2018-02-20:
### Version 1.39.1 b

I've just loaded up the latest beta version of Slic3r  and have noticed that the print speeds are really wacky compared to the MK2 in the current released version (1.38.6).
I also think that the 'Normal' and 'Fast' profiles are the wrong way around (normal is max speed and fast is normal)

Can someone confirm that these profiles are correct. I don't want to break something!
See attached screenshots
[image]

supermerill, 2018-02-21:
Sorry, but this git is only for slic3rPE, not for prusa profile. There are an other git project for that.
https://github.com/prusa3d/Slic3r-settings

Also, for questions, I think the forum (https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/) is a better place than a bug report database.

Don't forget to close this topic when your issue is resolved!

Disclaimer: I'm not a prusa employe, just a random guy on internet giving advice.

bubnikv, 2018-02-21:
We bundle the appropriate slic3r profiles for our printers now. You should run

[image]

[image]

Please note to check the "Reset user profile, install from scratch" checkbox.

After that, the profiles will look different.

It may be, that the profiles at https://github.com/prusa3d/slic3r-settings are not up to date.

bubnikv, 2018-03-14:
@chilternflyer Did the update of the profiles solve your issue?

chilternflyer, 2018-03-14:
Hi,
No it didn't really solve it.
However, the recently released version of it (1.39.1) has had its profiles tweaked and they are working very well. My Mk3 is now printing clear ABS better than my MK2S. So yeah I'm really happy with it!
You can close this issue now. Thank you for you help.

closed 2018-03-14 by chilternflyer