prusa3d/PrusaSlicer #730: Slic3rPE 1.39.1-beta Lithophane No layers were detected

TheZeroBeast, 2018-02-21:
Slic3rPE 1.39.1-beta

Mac OS X 10.13.3

### Behavior
* When using Inner Curve type for Lithophane @ "http://3dp.rocks/lithophane/" the model won't slice and presents error "No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry".

* _Steps to reproduce_ - Make any Inner Curve lithophane and try slice.

* _Expected Results_ - A sliced model to print which Slic3rPE 1.38.5 does with the same settings/configuration
* _Actual Results_ - Error as above

[IMG_3030.jpgW100H75T3V4B0A0C0PS.stl.zip](https://github.com/prusa3d/Slic3r/files/1743661/IMG_3030.jpgW100H75T3V4B0A0C0PS.stl.zip)
[image]
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-03-16:
This has been fixed with https://github.com/prusa3d/Slic3r/commit/efeda3dcdcd8fc83dbbe9c93e4a959c1254a68be
and it will be part of Slic3r PE 1.40.

closed 2018-03-16 by bubnikv

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.