prusa3d/PrusaSlicer #731: wipe at end of extrusion don't work

Hywelmartin, 2018-02-21:
### Version
1.39.1-beta-prusa-linux64

### Operating system type + version
Ubuntu 17.10

### Behavior
it doesen't backtrackat end of extrusion
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]
[image]
[image]
[slicer_stl_config.zip](https://github.com/prusa3d/Slic3r/files/1744667/slicer_stl_config.zip)

Hywelmartin, 2018-05-07:
ok tested in version_1.39.1

silly me

I taught that it would wipe the same distance anyway regardless how much I retract before

bubnikv, 2018-05-10:
Thanks for the report, Happily closing.

closed 2018-05-10 by bubnikv

nordurljosahvida, 2020-04-22:
@bubnikv i'm not sure i understand, does "retract amount before wipe" simply make the retraction happen earlier? or does it affect the amount retracted? what is the value that makes the retraction more "extreme", meaning reduces stringing the most?

Thanks

KMKAR, 2020-04-29:
@nordurljosahvida were you able to get some answer for the "retract amount before wipe" feature? Thanks

nordurljosahvida, 2020-04-29:
Hi @KMKAR, no I haven't received a reply yet, but I know things are busy around here so let's give it some time.

KMKAR, 2020-04-29:
@bubnikv would you have any available time to describe in detail the "retract amount before wipe" feature? I couldn't find documentation about it. Thanks