prusa3d/PrusaSlicer #732: Do not save profiles

blithe74, 2018-02-21:
### Version
Slic3r 1.39.1 beta prusa3d win64
Slic3r-1.39.1-beta-prusa3d-win64-full-201802131107.zip loaded from from github.com

### Operating system type + version
Windows 7 Professional x64 SP1 with all critical updates, Russia local.
The user without administrative rights.

### Behavior
The programm install on drive D: in dirrectory:
D:\3DPrinter\SoftWare\Slic3r-1.39.1-beta-prusa3d-win64-full-201802131107\
The Users dirrectory redirect from SSD to HDD (mklink  /J    C:\Users     D:\Users)

When click to icon "diskette" (with new or such profile name) the program do not save profiles. The dirrectory:
C:\Users\Internet\AppData\Roaming\Slic3rPE\filament\ 
C:\Users\Internet\AppData\Roaming\Slic3rPE\print\
C:\Users\Internet\AppData\Roaming\Slic3rPE\printer\
C:\Users\Internet\Application Data\Slic3rPE\filament\
C:\Users\Internet\Application Data\Slic3rPE\print\
C:\Users\Internet\Application Data\Slic3rPE\printer\
is empty.
The user have full access to directories (rights: Read, Write, Delete and etc. is enabled)
[image]

bubnikv, 2018-03-14:
Sorry, but the Windows reparse points (those made by mklink /J) are not supported well by the boost::filesystem library. I am not sure whether we should invest time into finding a workaround, as I don't expect many of the Slic3r users to hack around with the directory reparse points on Windows.

bubnikv, 2018-04-18:
See
https://svn.boost.org/trac10/ticket/10900
https://svn.boost.org/trac10/ticket/11873
https://svn.boost.org/trac10/ticket/11057
https://svn.boost.org/trac10/ticket/9016
https://svn.boost.org/trac10/ticket/5649

We will likely wait for the Boost guys to fix it.

blithe74, 2018-04-18:
Hi dear!
The original slic3r do not have this problem.
For solution a problem, You can assemble special version for me, which will show on display some debugging information, and  I  send  screenshots to you.

С уважением, Александр Шуковилов

bubnikv, 2018-04-18:
> The original slic3r do not have this problem.

This is interesting. I suppose the Prusa Slic3r has this code already
converted from Perl to C++.

Thanks for your offer, but I believe I can debug your issue myself.

blithe74, 2018-04-18:
The hint..., in original slic3r everything works properly. May be they used a another API function?
My offer for debuging stay as actual.

Excuse me for my English...

С уважением, Александр Шуковилов

chylek, 2018-11-18:
As a workaround you can launch slic3r.exe with the following argument: --datadir="pathWithoutSymlinks"
But if you have non-ASCII characters in that path it will fail again due to charset issues.

blithe74, 2019-02-22:
Hi dear!

Fan of cooling of the filament.

1) If for fan sets low speeds of rotation, there can be a situation when the fan cannot begin rotation. However if to push it, it will rotate. For reliable start a fan, I offer, at first moment set full tension, and then to reduce tension to a preset value.

2) For reliable gluing of a detail and preservation of a form I suggest to set 3 speeds of rotation: 
- speed for the print of 2 external perimeters;
- speed for the print of all another perimeters and infill;
- speed for print a bridges.
It will allow to make an equal and strong detail.

С уважением, Александр Шуковилов

bubnikv, 2019-02-22:
@blithe74 Hello dear!

> Fan of cooling of the filament.

It really does not belong to this thread. If you feel you need to discuss this topic, please check the existing issues first, if the topic fits there, please continue the discussion there. Otherwise open a new topic.

adras, 2020-03-30:
I just ran into the same issue #3956 . Looking at the date this issue was created I doubt, that the boost guys are going to fix it. Their issues are even older.

bubnikv, 2020-09-01:
Fixed with 683af51685ec21d09be4c0ef9ac8228b38bcb56d.

closed 2020-09-01 by bubnikv

bubnikv, 2020-09-01:
Fixed by not canonizing the symbolic links aka reparse points. The reparse points are not processed correctly by boost::filesystem::canonical() on Windows and making the path canonical (replacing symlinks with the path they point to) is not strictly required, therefore the fix does not take any functionality away.

I have tested both the directory symbolic links and directory junctions (mklink /d and mklink /s). Both seem to be processed correctly now by PrusaSlicer if used to point to PrusaSlicer user profile directory. I have also tested both types of these windows symlinks to point to PrusaSlicer/filament directory and it works as well.

The fix will be in the first public PrusaSlicer 2.3.0-alpha. Please let me know whether there is something yet broken in regard to windows symlinks.

blithe74, 2020-12-03:
Hi friends!
The Prusa Slic3r version 2.3.0.0 beta 1 have some problem (see attached files):
*  Crash
*  Strange work of support of model (in version 2.2.0 too) . See attached files.
Between the interface layer (dark green color)  and the supported layer of model (dark blue color): 3 layers of 0.2 mm = 0.6mm, instead 0.1 mm how in settings. For make  Z distance to 0.1mm You make huge (height) of first supported layer of model (dark blue color) and it to look very bad in product.
May be first supported layer to squeeze out at distance how in setting + heigh of layer with 100-110% of flow and full fan speed for to prevent sticking to interface layer?
 
 
With best regards, Alexandr.

bubnikv, 2020-12-04:
@ blithe74

Dear Alexandr.

Please don't mix issues. If you have any issues, please first search for similar ones (for example the support issues), or create a new one. Thank you.

> The Prusa Slic3r version 2.3.0.0 beta 1 have some problem (see attached files):

No file attached.

Does your report from yesterday has anything to do with
> Saving profiles into a Windows directory with reparse point fails (Boost::filesystem issue) #732
?