prusa3d/PrusaSlicer #734: Estimated time completely wrong for MM

Blizzke, 2018-02-22:
### Version
1.39.1-beta

### Operating system type + version
MacOS 10.13.3 (2nd release of that patch)

### Behavior
Did 2 multi-material prints ([this one](https://www.thingiverse.com/thing:2777345) and [this one](https://www.thingiverse.com/thing:2704415)) and for both estimated times where around 16.5hours. Both finished in around 5.5.

Blizzke, 2018-02-22:
[this one](https://www.thingiverse.com/thing:2333485) was estimated at 5h40 and printed in 2h10

FidelCapo, 2018-02-23:
Thank you for your report. I tried printing the first two objects (1Up and Mario), and the Slicer estimate was almost accurate. I add my configuration files and print times.

1UP - slic3r estimate: 14H9M - printed in 13H57M
Mario - slic3r estimate: 12H46M - printed in 12H36M

The reason why Slic3r shows you bad estimates can be in the wrong setting. I'm sending a link to the relase note dealing with this issue.
https://github.com/prusa3d/Slic3r/releases - Slic3r PE 1.39.1-alpha
Introduction of a new firmware type: Marlin
Slic3r always treated the RepRap, Sprinter and Marlin firmware types equally. However, due to the accurate print time estimator, starting with this release we had to separate the Marlin firmware from the crowd, as the Marlin firmware defines the M203 G-code (set maximum feedrate) in mm/sec, while Reprap and Sprinter define the same value in mm/min. If you are running Marlin, please make sure to switch Slic3r to a new "Marlin" firmware flavor to get a correct print time estimate.

Please understand, that the official Slic3r and the earlier versions of the Slic3r Prusa Edition do not support the new "Marlin" firmware flavor, therefore the "Marlin" firmware flavor will be reverted to the Reprap/Sprinter flavor, if you load the Slic3r PE profiles into the other Slic3r and save them.

Make sure you have set up in: Printer settings - General - Firmware - G-code flavor - Marlin and not RepRap Sprinter.

[PrusaMMprintGitHub734.zip](https://github.com/prusa3d/Slic3r/files/1751310/PrusaMMprintGitHub734.zip)

Blizzke, 2018-02-26:
It's weird that you can read something and not get it until it applies to you personally (I actually saw that piece of text before). Anyhow, switching to Marlin seems to result in a correct result. Thanks!

closed 2018-02-26 by Blizzke