prusa3d/PrusaSlicer #737: 1.39.1 Alpha No bridging speed after infill

erikcramerjr, 2018-02-24:
### Version
_Version of Slic3r used goes here_
Slic3r-1.39.1-alpha-prusa3d-win64-full-201802021436
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10  Version	10.0.16299 Build 16299
### Behavior
* _Describe the problem_
After infill the nozel doesnt slow down to do the infill.
With petg this result into blobs etc
* _Steps needed to reproduce the problem_
Create a print file with hollow space for instance lcd-supports.stl
after the infill on the first 10 layers the print head doesnt slow down to do the infill 
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_   the nozel slows down when doing the infill
* _Actual Results_
The nozel keeps the same speed and with petg this results into blobs
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
NO, Bug

[config.zip](https://github.com/prusa3d/Slic3r/files/1754309/config.zip)
[lcd-supports.zip](https://github.com/prusa3d/Slic3r/files/1754310/lcd-supports.zip)  (gcode + file)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

erikcramerjr, 2018-02-24:
Bug is also in 1.38,  mostly noticeble when the infill is not higher then 3 layers.
On higher infill the nozel does slow down.

bubnikv, 2018-05-10:
@FidelCapo Would you please verify this issue? Thanks.

bubnikv, 2018-05-16:
Would you please test with https://github.com/prusa3d/Slic3r/releases/tag/version_1.40.0-alpha1 ? We have a neat "by feature" G-code preview, this will help you to point us to what you believe is a problem.

I am just guessing, that what you are objecting about are the small magenta regions at he left bottom and right top. Is that correct?

[image]

[image]

bubnikv, 2018-08-02:
It is unclear what the issue is supposed to be, no response.

closed 2018-08-02 by bubnikv