prusa3d/PrusaSlicer #740: Hanging Support

fatalelieberi, 2018-02-25:
### Version
Slic3r PE 1.39.1-beta-prusa3d-win64

### Operating system type + version
Windows 10 Pro 64Bit Version 1709

### Behavior
When slicing an object with curves and a slot in the curve, the support is generated outside the object and it's in the air. The support shouldn't be there, instead it should be on the flat surface in the slot. 

To reproduce it, slice the attached stl file, with support activated (without ticking "support on build plate only" option).

### STL, Configuration & Screenshot
PrintSetting - [PrintSettingConfig.zip](https://github.com/prusa3d/Slic3r/files/1755909/PrintSettingConfig.zip)

STL file - [ArcReactorAsm01_P17009-003-Base_2.zip](https://github.com/prusa3d/Slic3r/files/1755913/ArcReactorAsm01_P17009-003-Base_2.zip)

Support in the air
[image]

bubnikv, 2018-02-26:
Actually these supports are technically not hanging in the air, they are supported by the support base, which is sparse. It is just the way the Slic3r PE projects the supports into a fixed cell size grid, which makes the supports to to protrude outside the overhang.

Simplify3D uses the same strategy as Slic3r PE: it projects the supports into a fixed grid. Cura or kisslicer do it differently, they try to only place the supports below the overhangs. Both strategies have their pros and cons (Slic3r PE makes the supports more stable, while Cura saves material and creates less scars on the top surfaces), but I agree we should improve Slic3r.

bubnikv, 2021-03-22:
Testing snug supports

[image]

[image]

bubnikv, 2021-03-23:
Fixed with af9c7c9

    Implementing a new switch for the shape of support towers:
    expanded to a grid (the old way) vs.
    snug (like the upstream Slic3r, Cura or Ideamaker).
    
    Snug supports suffered from the degeneracies when merging overhang islands
    over a large number of layers when projecting the support towers down.
    We borrowed the idea & a bit of code from Cura by simplifying the support
    polygons by closing the concave cracks, see the smooth_outward() function
    and the MutablePolygon class.

closed 2021-03-23 by bubnikv