prusa3d/PrusaSlicer #743: FR - Conditional GCode and Variable documentation

gregspruce, 2018-02-26:
I'm trying to create tool change scripts to better deal with oozing by ram-purging into a purge bucket.

I'm looking for documentation or examples of how to use slic3r's variables and conditional gcode. For example, setting temperatures in my tool change code. It seems as if the print temps are an array, but what formatting can I use to call it properly in my script? For example, I can see in the end of a .gcode file that first_layer_temperature = 220,235. If I want to use the T0 first layer temp, is that {first_layer_temperature(0)}?

Thanks for any help.

bubnikv, 2018-02-26:
I commented on the Prusa3D forum:

> I am happy people are finding good use of the new macro functionality. As of now there is no documentation. In general, you may reference any variable you will find in the config file. The extruder dependent values are vectors because Slic3r supports multiple extruders. Currently the macro language supports the {if condition}{elsif condition}{else}{endif} syntax, where the condition is an expression in a syntax derived from the C language, with the addition of a Perl language like regex expression.
> 
> The macro processor also supports {expression}, which simply evaluates and places the result of an expression into the G-code. You may again use the usual C expression syntax including the ternary operator.
> 
> You will find the grammar definition as part of the source code, starting with line 880:
https://github.com/prusa3d/Slic3r/blob/ ... r.cpp#L880
up to line 1043, so it is not really complex.
> 
> You are most welcome to contribute your findings and use cases to the Slic3r PE wiki :-)
> https://github.com/prusa3d/slic3r/wiki

And Francesco really contributed the following guide:
https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language

I hope you will find it useful, and you may even contribute to it :-)

bubnikv, 2018-03-14:
We have a wiki page.
https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language
Closing.

closed 2018-03-14 by bubnikv