prusa3d/PrusaSlicer #744: Missing support interface on low contact Z distance [regression]

comps, 2018-02-26:
### Version
1.39.1-beta1 (precompiled linux64 binaries)

### Operating system type + version
Linux, Fedora 27

### Behavior

Slic3r PE version 1.34.0, I'm guessing commit 40b75f6cee3d, introduced a new interface calculation that breaks support interface for contact Z distance below layer height. Using such low values, particularly near-0 distance **while still using bridging flow** yields the best printing results in my experience, particularly with high-detail miniatures.

The "Contact Z distance" in Slic3r works like this (when above zero, as zero disables bridging flow):

[image]

This shows the typical 0.2 "breakable" supports distance. Slic3r PE 1.33.8 however allowed values much lower, ie. 0.001 (which was >0, so it used bridging flow, but still very low to achieve good quality):
[image]
(The above was determined from analysis of the generated gcode.)

For a practical demonstration, see these photos:

[image]

The problem and reason why I can't use 0.001mm in newer versions is that Slic3r PE >= 1.34.0 does not create interface when the contact distance is this low:

[image]

There seems to be some "magical" number around 0.06mm below which this behavior happens (when using 0.2 layer height).

#### STL/Config (.ZIP) where problem occurs
[stl-with-config.zip](https://github.com/prusa3d/Slic3r/files/1759789/stl-with-config.zip)

bubnikv, 2018-03-14:
> There seems to be some "magical" number around 0.06mm below which this behavior happens (when using 0.2 layer height).

Isn't the "magical" number the minimum layer height defined for the extruder at the printer profile->extruder page? This value came with the smooth variable layer height editing, and its purpose is to avoid extruding unrealistic thin layers.

comps, 2018-03-15:
> Isn't the "magical" number the minimum layer height defined for the extruder at the printer profile->extruder page? This value came with the smooth variable layer height editing, and its purpose is to avoid extruding unrealistic thin layers.

Thanks for pointing that out - yes, that seems to have an effect on it. There is a limit (contact z distance below 0.01 causes the interface to be ignored even if I set minimum layer height to 0 or 0.001), but the setting definitely has an effect.

To state the obvious, the issue here is that this isn't about minimum layer heights, but about support-to-model Z distance separation (which itself isn't an extrusion).

bubnikv, 2018-03-16:
I suppose your report is a duplicate of one of the following: https://github.com/prusa3d/Slic3r/labels/supports

Most likely https://github.com/prusa3d/Slic3r/issues/102 and https://github.com/prusa3d/Slic3r/issues/612

May I close this issue as a duplicate?

comps, 2018-03-16:
> Most likely #102 and #612

Reading through them, the issue seems similar, but not really the same. However if you believe a possible fix would be the same for both, sure, close the issue.

closed 2021-01-20 by rtyr