prusa3d/PrusaSlicer #745: [Request] Export G-Code to /dev/null on Slice

CapnBry, 2018-02-27:
### Version
Slic3r-1.39.1-alpha-prusa3d-win64-full-201802021436

### Operating system type + version
Windows 10 Pro x64

### Behavior
The Sliced Info box on the platter tab is great and provides a lot of useful information, but it is only generated when G-Code is output. This means that for tweaking settings in Slic3r, you have to "Export G-code...", select the file, Yes overwrite each time you want to see how much of a difference changing something in the profile changes print time or material usage.

It would be great if when Slicing occurs, the G-code generator would always run even if no file is being written or sent to OctoPrint. This can be as simple as slicing to the standard temp file if no other destination is pending (in `lib/Slic3r/GUI/Plater.pm::on_process_completed()`)
```
    return if $error;
    $self->{toolpaths2D}->reload_print if $self->{toolpaths2D};
    $self->{preview3D}->reload_print if $self->{preview3D};

+    $self->{export_gcode_output_file} = Wx::StandardPaths::Get->GetTempDir()
+        if (! defined($self->{export_gcode_output_file}) && ! defined($self->{send_gcode_file}));

    # if we have an export filename, start a new thread for exporting G-code
    if ($self->{export_gcode_output_file}) {
        @_ = ();
```

It would be better if `void GCode::do_export()` supported a nullptr path which would then not open a file at all in its function body. The only changes needed there would be to guard against trying to fflush() and fclose() the file, and `void GCode::_write(FILE* file, const char *what)` would not fwrite() if nullptr == file.

I think having the Sliced Info always appear after slicing instead of after G-code export would be much more intuitive. It is labeled *Sliced* Info not G-code Info (even if it is completely based on the G-code, not the slicing). This only requires a few lines of code and not much CPU time.

#### STL/Config (.ZIP) where problem occurs
All

bubnikv, 2018-03-14:
Thanks for your proposal. We are looking around for a solution to run the G-code generator at the end of the slicing process automatically.

bubnikv, 2018-08-02:
We are reworking the background processing to run the G-code export at the background into a temp file (or memory). Closing for now.

closed 2018-08-02 by bubnikv