prusa3d/PrusaSlicer #748: New feature request: Flow control tune

blithe74, 2018-02-28:
This a new feature request.

Kind health for You,  friends!
Thank you for your work!

Your program not bad prepare model for print at small speeds, but if to establish print speed faster then 80 mm / sec, the result turns out extremely unsatisfactory. By means of the available settings in "Advanced settings" I can make a certain elements of a detail  is ideal, but not for all detail in general. 
It is caused by specific features of the printer, the filament, a design of a nozzle, print speed, humidity of the filament and humidity indoors, etc. It cannot be described by one general algorithm.
I offer, except the general adjustment of amount of a plastic (flow), in the bookmark "Advanced Settings" to make  individually adjustment a flow for each element of a print: Perimeters, Infill, Solid, Gaps, Bridges, etc. It will allow to achieve excellent quality of a seal of all detail.  
Besides, in my opinion, it will be more convenient if instead of "the extrusion width" make parameters: "the step of threads" and "a flow (amount of the squeezed-out plastic)". By default width (for flow = 100%) )will be calculated from formula:
(3.14159 * High_Layer^2 / 4 + Diameter_Nozzle * High_Layer) * Len_layer = 3.14159 * Diameter_Filament^2 / 4 * Len_Filament

Yours faithfully Alexander.

Sebastianv650, 2018-02-28:
The effect your are seeing is no static thing which could be adjusted by a dedicated flow factor. It's a dynamic pressure which needs to be compensated, and the slicer is not the right place to do it as it doesn't know all the parameters.
There is an advanced feature called LIN_ADVANCE which is available in Marlin and also PrusaFW (latest version not yet integrated), you can find details about that here: http://marlinfw.org/docs/features/lin_advance.html

> Besides, in my opinion, it will be more convenient if instead of "the extrusion width" make parameters

I don't understand what you want to say I guess. But personaly I need to know which line widths my prints uses, therefore everything else than real mm values wouldn't be acceptable for me.

bubnikv, 2018-03-14:
> I don't understand what you want to say I guess. 

I don't understand the proposal either. Would you please expand on that?

bubnikv, 2018-04-18:
Unclear proposal, no response, closing.

closed 2018-04-18 by bubnikv

blithe74, 2018-04-18:
I apologize that so long couldn't respond to your last letter.
I understand that the algorithm of calculation of amount of plastic is very difficult. But You can't consider the fluidity and parameters of glue for each plastic. With certain values of a stream and step I printed ideal filling of the top layer, but at the same time the perimeters and the infill had a bad filling. If I printed the ideal perimeters, I had a more extrusion on the top layer. Therefore I offer except the main flow control, to add adjustments a flow control for top fill, Infill, Internal perimeters and external perimeters.

Couldn't you send to me optimum settings of print for speed - 80 mm/s, Nozzle dia - 0,4 mm, filament 1,75 mm. Also .gcode file with the photo of result that it was possible to compare.
With best regards Alexandr.

С уважением, Александр Шуковилов