prusa3d/PrusaSlicer #749: Feature Request: Make last layer_z printed available in the end print gcode processing.

HeySideburns, 2018-03-01:
It would be nice to be able to access the last layer_z placeholder value in the end print custom gcode so a user could detect if their print head is below a certain threshold, then raise it to that threshold. In my case it is so it will always raise to give me clearance to lift off my magnetic bed. However, above certain heights this movement is unnecessary. My use case would need the last layer_z that was used, but it may also be useful for people to have access to the highest layer_z used as well. Looking at the code it seems it is possible for those to be different if each object is printed separately.

bubnikv, 2018-05-10:
This should be reasonably easy to do. Let's see whether we will manage it for the upcoming 1.40 release.

bubnikv, 2018-05-16:
Implemented with https://github.com/prusa3d/Slic3r/commit/7645e9cb7a884a4722324b7b5a839424291299fb

Added layer_num, layer_z variables to the end_gcode, end_filament_gcode, added min(.,.), max(.,.) functions to the placeholder parser syntax.

This allows one to sketch an end g-code like this to lift the print head at the end of the print:

```
; Move the print head up
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+20, max_print_height)}{endif}
```

bubnikv, 2018-05-16:
This change will be part of the upcoming 1.40.0-beta.

closed 2018-05-16 by bubnikv

vertigo235, 2018-06-04:
I tried this and it didn't work.  

Is max_print_height a variable or am I supposed to replace that with some value like 210?

bubnikv, 2018-06-04:
> I tried this and it didn't work.

Would you please provide your configuration?

> Is max_print_height a variable or am I supposed to replace that with some value like 210?

Both should work.