prusa3d/PrusaSlicer #750: Print Time Estimator not working right.

LaserBee, 2018-03-02:
### Version
Slic3r for Prusa 3.19.1 Beta Win64

### Operating system type + version
Windows 10

### Behavior

Until a few days ago, the print time estimator was nearly bang-on. A couple minutes out over the course of an hour. Brilliant. However, something's changed, and now my print times are way off. My latest print, Slic3r said it would be 4h04m14s, and the final time is nearly 4:40. I am using an Anet A8, running Marlin, if that's important. Note, I did follow the instructions and add the following code to the start gcode box in printer settings:

>M201 X2000 Y2000 Z100 E10000
>M203 X400.00 Y400.00 Z8.00 E50.00
>M204 P400.00 R1000.00 T1000.00
>M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00

I've tried re-installing Slic3r, put the acc and retraction speeds back into the gcode, &c. Nothing seems to have helped!
_Is this a new feature request?_

[STL and config ini.zip](https://github.com/prusa3d/Slic3r/files/1773289/STL.and.config.ini.zip)

bubnikv, 2018-03-02:
most likely your printer type is set to reprap, while it should be set to
marlin.

LaserBee, 2018-03-02:
Thanks, Bubnikv, but it's definitely Marlin!

bubnikv, 2018-03-02:
The difference between 4h04m and 4:40 is not that big. I suppose there is a difference between the Anet's Marlin and our Marlin fork. Also if you have a too detailed print, than the printer may not keep up and the real print time will be larger than the estimated print time, as the Slic3r's time estimator does not simulate firmware choking.

LaserBee, 2018-03-02:
It's strange though, as it was working perfectly well last week!

bubnikv, 2018-03-02:
>  It's strange though, as it was working perfectly well last week!

It is hard to say. You may have changed something on your configuration, or
it is just G-code dependent, for example

1) Too many small segments will choke the firmware, especially if there is
a graphic controller connected to the Arduino board, or if printed over
OctoPrint.

2) There is a mismatch in feed rates / acceleration / jerk settings between
what Slic3r thinks and what the firmware uses. Then often the difference is
the largest for long moves, where the firmware has chance to accelerate to
a full speed.

LaserBee, 2018-03-02:
Well, thank you very much for your help. I might re-flash Marlin,
re-install Slic3r, and re flash Octoprint. Go back to the start!

coreyward, 2018-03-05:
Hmm, I'm having similar discrepancies (2:54 vs estimated 2:37, or 11.5%) with a stock MK3 running the latest firmware (2.1.4) on a simple model. I'd consider this great, but Josef has been touting an error rate “way below 5%” for this feature.

bubnikv, 2018-03-05:
@coreyward The print time estimate does not contain the bed / nozzle preheat times, that may easily add 5 minutes to ABS prints. If you find any print with larger print time estimate discrepancy, please share the data with us and we will check that.

coreyward, 2018-03-05:
@bubnikv This was a PLA print and I was preheated.

bubnikv, 2018-03-14:
@coreyward If you find any print with larger print time estimate discrepancy, please share the data with us and we will check that.

bubnikv, 2018-03-14:
Guys, please provide sample data, otherwise I would have to close this issue without resolution. Thank you.

LaserBee, 2018-03-14:
Hi, sorry, what kind of data should I provide?

bubnikv, 2018-03-14:
A config file (go to file->export config), an STL, the G-code, the time
estimated and the actual time shown by the printer. All this data is
required for us to reproduce your scenario in house and track the possible
issues. Best if you are running our printer :-), then we can print the
G-code while watching the printer, otherwise we could at least analyze the
data provided, whether it contains something suspicious.

LaserBee, 2018-03-14:
Okay, thank you. I am running an Anet A8, so it's *almost* like one of
yours. :)

I am planning a big print in the next couple days, can I give you the data
for that? At he moment, I am only printing half-hour things, which I don't
believe will demonstrate the problem very well...

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bubnikv, 2018-03-14:
@LaserBee Well, the time estimator simulates our version of the Marlin firmware, so there may be some differences. We would rather tune the time estimator against our machines, as we have a full control over the parameters. With a 3rd party printer, you may have some differences in the firmware (some limits, CPU overload etc), which we have no way (and resources) to troubleshoot.

LaserBee, 2018-03-14:
Yup, I understand that. If you'd like to close the ticket, that's cool. The
estimate seems to be closer again to what it was. I suspect it's my cheap,
chinese knock-off printer that sometimes just decides to take it's time. :)

Thanks for all your help (and an amazing product!)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bubnikv, 2018-03-15:
@sepotvin Thanks, @FidelCapo will verify your print.

LaserBee, 2018-03-17:
Hi again. Here's the data I can provide you. It was only an XYZ test cube,
but Slic3r estimated 22:30, and the actual print time was 29 minutes. Both
the bed and the nozzle were pre-heated, so that is actual print time.

Gcode, stl, config.ini incuded and Marlin Configuration.h files are
included.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ManuelMcLure, 2018-03-17:
I don't know if this will help since it's not a Prusa machine, but this file was estimated at 12:59:33 and took about 14:50 to print.
Machine is physically an Anet A8 with the motherboard replaced with an Arduino 2560 + RAMPS 1.4, running bog-standard Marlin 1.1.8.
[PrintTime.zip](https://github.com/prusa3d/Slic3r/files/1821905/PrintTime.zip)

Sebastianv650, 2018-03-17:
Just for info, current marlin has a similar, but most likely not equal planner compared to prusafw.

bubnikv, 2018-03-18:
@LaserBee You did not manage to attach the files.

> It was only an XYZ test cube, but Slic3r estimated 22:30, and the actual print time was 29 minutes.

Such a huge error most likely indicates an incorrect Slic3r configuration.

@ManuelMcLure

> I don't know if this will help since it's not a Prusa machine, but this file was estimated at 12:59:33 and took about 14:50 to print.
> Machine is physically an Anet A8 with the motherboard replaced with an Arduino 2560 + RAMPS 1.4, running bog-standard Marlin 1.1.8.

That is about 11% of time difference. Unfortunately you test on a machine, which we do not own, so we cannot verify your experiment. I would rather wait for data indicating discrepancies on our machines, where we could pin-point the cause easier.

LaserBee, 2018-03-18:
I've double checked that I have put those acceleration, jerk, &c settings
in correctly. I have the four lines at the top of my starting gcode. What
other configurations might contribute to this?

bubnikv, 2018-03-18:
>  What other configurations might contribute to this?

Do you have the firmware flavor set to Marlin in Slic3r PE?

LaserBee, 2018-03-18:
Yup! Firmware is marlin. And I haven't changed anything else really except
the usual temperature and layer settings.

FidelCapo, 2018-03-20:
@sepotvin 
I tried to print your gcode on our MK3 printer and the total print time was 15H 44M, which is a bit less than the estimate of your slicer which was 16H 10M 33S. Did you print from an SD card or, for example, over an octoprint? In prints over octoprint, there may be differences in estimated time and real-time printing. Did you try using a different SD card?

FidelCapo, 2018-03-20:
@ManuelMcLure I tried to print your STL with your configuration on our MK2 printer and the total print time was 13H 3M, which is a bit less than the estimate of slic3ru which was 12H 59M. Unfortunately, I do not have the opportunity to test this case on the same printer as yours (Anet A8).

LaserBee, 2018-03-20:
Another attempt to attach the files!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ManuelMcLure, 2018-03-20:
> @ManuelMcLure I tried to print your STL with your configuration on our MK2 printer and the total print time was 13H 3M, which is a bit less than the estimate of slic3ru which was 12H 59M. Unfortunately, I do not have the opportunity to test this case on the same printer as yours (Anet A8).

Fair enough - I'll have to see if I can figure out what's slowing things down on my printer. It could be OctoPrint since that's what I use to control the printer. I'm not getting any skips/layer shifts so I don't think it's that the printer can't physically achieve the speeds/accelerations specified in the file, but I don't notice any stuttering either.

bubnikv, 2018-03-20:
It is quite possible, that OctoPrint does not quite keep up with the
printer. In that case the Marlin firmware slows down the print artificially
to avoid stuttering. As long as OctoPrint keeps up somehow, the printer
slow down is smooth, therefore not obvious.

sepotvin, 2018-03-21:
@FidelCapo 
I'm printing via the SD Card that came with the printer. I tried with a different one and got the same result. The only setting I changed (except for the Z tuning) is the Stealth mode. Would using Stealth mode increase the print time?

bubnikv, 2018-08-02:
Stealth mode support has been implemented in Slic3r 1.40.0-alpha3. The new Slic3r works in tandem with the upcoming MK3 firmware. Also we have changed the MK3 stealth mode limits, so it runs just a bit slower than the normal mode.

Closing.

closed 2018-08-02 by bubnikv