prusa3d/PrusaSlicer #751: [Request] *support* generation threshold area

BETLOG, 2018-03-02:
### Version
Slic3r-1.39.1-beta-prusa3d-linux64-full-201802131106

A *support generation threshold area* would be useful.
Similar to "Solid infill threshold area", but for preventing infill from populating small spaces.

eg: When using "support on build plate only" is not an option, but I do not want my bolt-holes filled with plastic.
[image]

zoff99, 2018-04-13:
also support is outside of the object (in X and Y direction). like on the image above.
very strange. sometimes it will create support on top of a top layer, which can not even work.

bubnikv, 2021-03-18:
Similar to #1928
Also you can now block the supports using support blockers or paint-on supports.

closed 2021-03-18 by bubnikv