prusa3d/PrusaSlicer #752: Auto zoom is annoyingly 'eager' and too wide

BETLOG, 2018-03-02:
### Version
Slic3r-1.39.1-beta-prusa3d-linux64-full-201802131106

### Operating system type + version
kubuntu 14.10

### Behavior
- Auto-zoom is annoying. 
  - It does not need to happen after every single manipulation.
  - *It is particularly annoying and counter-intuitive when engaged in "layer Editing" mode.*
- Auto-zoom can not be disabled. 
  - It should be a preferences option.
- Auto-zoom is too wide/too far away. 
  - There is too much empty space left around the plater/objects after zoom... camera is too far away.

* _Expected Results_
Auto-zoom should be an option.
Zoom should tightly frame the objects/plater.

* _Actual Results_

Zoom and rotate camera to the desired viewpoint. (assume it's a relevant fairly specific POV and annoying to repeat)
[image]

Moving an object *at all* triggers auto-zoom.
[image]

_Is this a new feature request?_
Maybe. Options include:
- modify 'eagerness' of autozoom conditions.
- add optional on/off checkbox in preferences or main gui.

bubnikv, 2018-03-14:
What do you mean with auto zoom? Do you mean auto-centering of an object to the center of the build plate? You can disable that in the application preferences.

BETLOG, 2018-03-15:
I do not mean "auto-center parts", I already have that disabled.

[image]
Open slic3r, default zoom

[image]
Drag&drop object, viewport automatically zooms to encompass object.
Larger objects get progressively worse zoom levels, an object that fills the build plate will be auto-zoomed so that there is more space around the object than there is of the object in the viewport. The image above is a good ratio of zoom, apparently because it is a tiny object.

[image]
Manually zoom in to desired feature...

[image]
Grab and move object, or do any number of other transforms or layer edits..... 

[image]
Auto-zoom kicks in and recomposes the viewport.
Again, this is a good zoom ratio, normally they get progressively 'wider' (less object and more empty space) as the object size increases.

[image]
Drag&drop a 210x210 mesh object...
Automatic zoom activates.
Note the very wide zoom, and object to empty space ratio.

Bad zoom ratio.,and automatic zooming are undesirable as currently enforced.
 'Auto' needs to be optional.
'Ratio' needs to be tight around the model, with minimal empty space.

BETLOG, 2018-03-30:
I only just now realised I forgot to mention that.
I concur that it is related, and anomalous.

dartrax, 2018-04-03:
Today I got this ;)

[image]

enricoturri1966, 2018-04-23:
This issue should be fixed by: https://github.com/prusa3d/Slic3r/commit/f8b1dc550685a8ca76c4d7a810151eabf08db2c1

bubnikv, 2018-05-10:
Fixed in 1.40 alpha. Stay tuned for the upcoming release.

closed 2018-05-10 by bubnikv