prusa3d/PrusaSlicer #755: Error when trying to start Slic3r Prusa Edition on CentOS

Linuxmonger, 2018-03-04:
### Version
1.39.1 beta 1

### Operating system type + version
CentOS Linux release 7.4.1708 (Core)  
[steve]$ uname -r
3.10.0-693.11.1.el7.x86_64

### Behavior
After download and extraction of;
Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542.tar.bz2

[steve@betty Downloads]$ tar -xjf Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542.tar.bz2 
[steve@betty Downloads]$ ./Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/slic3r
Can't load '/home/steve/Downloads/Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /home/steve/Downloads/Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so) at /home/steve/Downloads/Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/lib/site_perl/5.22.0/XSLoader.pm line 70.
 at /home/steve/Downloads/Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/lib/site_perl/5.22.0/Slic3r/XS.pm line 22.
Compilation failed in require at /home/steve/Downloads/Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/lib/site_perl/5.22.0/Slic3r.pm line 49.
BEGIN failed--compilation aborted at /home/steve/Downloads/Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/lib/site_perl/5.22.0/Slic3r.pm line 49.
Compilation failed in require at ./Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/slic3r.pl line 16.
BEGIN failed--compilation aborted at ./Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542/slic3r.pl line 16.
[steve]$ tar -xjf Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542.tar.bz2 
[steve]$ mv Slic3r-1.39.1-beta1-prusa3d-linux64-full-201802221542 ~/Slicer
[steve]$ ./Slicer/slic3r
Can't load '/home/steve/Slicer/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /home/steve/Slicer/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so) at /home/steve/Slicer/lib/site_perl/5.22.0/XSLoader.pm line 70.
 at /home/steve/Slicer/lib/site_perl/5.22.0/Slic3r/XS.pm line 22.
Compilation failed in require at /home/steve/Slicer/lib/site_perl/5.22.0/Slic3r.pm line 49.
BEGIN failed--compilation aborted at /home/steve/Slicer/lib/site_perl/5.22.0/Slic3r.pm line 49.
Compilation failed in require at ./Slicer/slic3r.pl line 16.
BEGIN failed--compilation aborted at ./Slicer/slic3r.pl line 16.

This is reproducible every time, and the same error occurs when using the appimage version, though the line numbers are different.

The error occurs with or without a .stl file, and as of yet, I don't know what the config is.

bubnikv, 2018-03-05:
I suppose this is the same issue as https://github.com/prusa3d/Slic3r/issues/182
We would need to build the AppImage on an older system than we currently do.

Linuxmonger, 2018-03-05:
This is the current version of CentOS/RHEL/etc... 
I thought the point of the AppImage packaging was to avoid this sort of issue.
Can you tell me what it is that I need to update to run the software?

vojtechkral, 2018-03-05:
This issue is documented in AppImage docs: https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#libstdcso6

Basically we need an older libstdc++ and according to [a similar issue in unrelated software](https://github.com/FoldingAtHome/fah-issues/issues/1147) we need a version older just by one patch version (`libstdc++.so.6.0.20` vs `libstdc++.so.6.0.19`).

Looking into it further...

jreybert, 2018-08-08:
Do you think it is fixable? According to the wiki, `gcc 4.9` is a hard requirement:

> Slic3r requires a C++ compiler that suppports most of the C++11 spec. For GCC, this means 4.9 or newer is required (this is a hard requirement). Clang 3.3 should also work (and is used on OSX).

Not sure it is possible to find a configuration with `gcc 4.9` **and** `libstdc++.so.6.0.19`.

schneis, 2018-10-12:
Any possibility of a fix, or anything that can be done to assist with a fix?  I am also running CentOS 7 and am trying to get Slic3r (v1.41.1) to work.

I tried compiling from source, but the default cmake (v2.8.12.2) is also too old so am having to put it on hold for the night.

vojtechkral, 2018-10-15:
@schneis I hear there's CMake 3.x available on CentOS 7 through EPEL, package named `cmake3`. Perhaps that could be a way of building Slic3r. (I'm sorry I can't provide better help than that at the moment.)

schneis, 2018-10-17:
@vojtechkral - Thank you.  I installed cmake3 and got a bit further, but am still running into problems.

Currently cmake is failing because one of its modules is not found:
```
CMake Error at /usr/share/cmake3/Modules/FindBoost.cmake:2044 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.53.0

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_log

  Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  xs/CMakeLists.txt:571 (find_package)
```

cmake is also failing because it can't find AlienWx.  Building that is running into some other issues that I'm unable to dig into right now.

jreybert, 2018-10-17:
I am not sure what you intend to do, but as gcc 4.9 is a hard requirement, meaning libstdc++.so.6.0.19 is also a hard requirement. In the end, once you will have face the dependencies issues, you will be blocked because of gcc 4.9 requirement.

vojtechkral, 2018-10-17:
@jreybert @schneis I believe that problem can be solved by installing one of the developer toolset packages, no? See, for example, [Developer Toolset 7](https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/)

In fact, I should've suggested this before suggesting `cmake3`. Sorry about that. Unless I'm wrong, Slic3r could be used through the devtoolset, although it's still not super convenient - you need a devtoolset-enabled shell to run the app.

jreybert, 2018-10-17:
I am not sure. It seems that devtoolset provides `devtoolset-7-libstdc++-devel.x86_64`, but not `libstdc++` itself. Once compiled, a hello world example points to standard libstdc++.
```
$ g++ -std=c++11 hello.c -o hello-gcc-std 
$ ldd hello-gcc-std
	linux-vdso.so.1 =>  (0x00007fffc65be000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007efc5d4ca000)
	libm.so.6 => /lib64/libm.so.6 (0x00007efc5d1c8000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007efc5cfb2000)
	libc.so.6 => /lib64/libc.so.6 (0x00007efc5cbe5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007efc5d7d1000)
$ scl enable devtoolset-7 bash
$ g++ -std=c++11 hello.c -o hello-gcc-std 
$ ldd hello-gcc-dev
	linux-vdso.so.1 =>  (0x00007ffcd5191000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f3276424000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f3276122000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f3275f0c000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f3275b3f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f327672b000)
```

vojtechkral, 2018-10-18:
@jreybert Looks like the devtoolset only has a static libstdc++ , or even just additions on top of the system lib.

Honestly, CentOS is pretty infuriating...

jreybert, 2018-10-18:
If it provides a static libstdc++ (which would make sense with the -devel rpm), we have a little chance to build libboost statically (not sure it is possible), cross fingers that no other lib have dependency on recent libstdc++, and maybe we should build a slic3r with a static libstdc++ (which is the final goal here).

I personally gave up, and use slic3r from a Windows VM...

vojtechkral, 2018-10-19:
@jreybert You could probably build some/all slic3r dependencies statically in the devtools environment using my Makefile in `doc/deps-build/unix-static/` (It's commented) and then build Slic3r against that. I think it should be possible to build Slic3r on CentOS7 one way or another.

The thing I'm more worried about - and a bit aggravated by - is that if we (the Slic3r team) want to support & build for CentOS/RHEL 7 we're basically forced to use CentOS7 as the build machine.

schneis, 2018-10-24:
Ok - thank you ... this is apparently much further over my head than just recompiling.  

I've installed a Windows VM to be able to slice & print for the time being.  I would really like to be able to run Slic3r on my CentOS 7 Desktop so that I am not dependent on Microsoft.

rtyr, 2020-10-20:
Is this issue still valid? Can we close it?

Linuxmonger, 2020-10-21:
Just based on age, I have to say, yes, please close this ticket.
Thank you!

closed 2020-10-21 by rtyr